Ultimaker/Cura #11043: project name totally changes after a save

rdbischoff, 2021-12-09:
### Application Version

4.12.0

### Platform

MacOS 10.14.6

### Printer

Ender 3 Max

### Reproduction steps

Set up project, slice, save. project name changes. I don't know what's happening. It often goes to a condensed version of a previous saved project.

### Actual results

Sample of what happens. 1) STL imported 2) Sliced 3) Saved

[image]
[image]
[image]

### Expected results

for the project name to stay the same.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I don't know what else to add.

fieldOfView, 2021-12-09:
You did not add a log file, and without it we cannot even begin to diagnose what is going on.

If I had to guess, I would say you have a plugin installed that does this. Gcode Filename Format (Plus)?

rdbischoff, 2021-12-09:
Where do I get a log file?

fieldOfView, 2021-12-09:
https://github.com/Ultimaker/Cura/#logging-issues

Ghostkeeper, 2021-12-22:
Since Cura 4.12 we're updating the print job name to the file name you give it when you save a project. This was deemed better for UX. Of course, every change breaks someone's workflow.

But your expected results are not my expected results. I'd expect the print job name to update to however you named the file. (How you named the file could be seen in those logs, if you please.)

rdbischoff, 2021-12-23:
I'm looking in my application support folder and there is no Cura subfolder for the logs.

fieldOfView, 2021-12-23:
Go to Help -> Show configuration folder in Cura. The logs should be in the folder that opens.

rdbischoff, 2021-12-23:
[cura.log](https://github.com/Ultimaker/Cura/files/7770606/cura.log)

fieldOfView, 2021-12-23:
I see that you have the "Printjob Naming" plugin installed. Try going to the Installed tab of the Marketplace, and disable that plugin. If the issue is fexed then, you may want to log an issue over at the Printjob Naming plugin issue queue:
https://github.com/fieldOfView/Cura-CustomJobPrefix/issues

rdbischoff, 2021-12-24:
ok. will do. it appears that it is that plugin, but i really like the plugin so I will just work around it until it gets fixed. I like having the printer added onto the end of the file automatically since I have an Ender 3 Pro and an Ender 3 Max.