Ultimaker/Cura #11046: Cannot slice models taller than ~550mm

colin-donar, 2021-12-09:
### Application Version

4.12

### Platform

Windows 10

### Printer

Custom pellet printer

### Reproduction steps

1. Create printer profile with Large Z height
2. import large model
3. slice

### Actual results

4. If model is larger than ~550mm then cura wont slice anything and the preview will be blank. see image full.png

[image]

### Expected results

Sinking the model into the bed to shorten the sliced model will result in cura correctly displaying the gcode preview. see image short.png

[image]

both images are taken after clicking "slice"

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have replicated the issue on multiple machines (including machine with activated windows)
I have found Version 4.9 does not have this issue...

I can provide log/project files eventually if requested. Currently away from that computer...

fvrmr, 2021-12-10:
Hi @colin-donar thank you for your report. 
I need indeed your log and project file to see what is causing this. You can share this in a .zip here on Github.

colin-donar, 2021-12-10:
[cura issue.zip](https://github.com/Ultimaker/Cura/files/7693772/cura.issue.zip)
Log and Project file

colin-donar, 2021-12-16:
Any update @fvrmr?

Ghostkeeper, 2021-12-22:
It's currently working on master and 4.13. I believe this is fixed by https://github.com/Ultimaker/CuraEngine/commit/75d826437aaa18a27c65a18f5620ae960e6511ba, after a similar report on Reddit. This fix is included in 4.13. Could you try with the beta?

closed 2021-12-22 by Ghostkeeper

colin-donar, 2021-12-22:
Just tested the beta and the bug is gone! Thanks!