Ultimaker/Cura #11048: Optional Serial Slicer

mattwn, 2021-12-10:
### Is your feature request related to a problem?

Interlayer travel moves increase print time and are a source of reduced print quality if stringing exists or combing does not work as expected. Interlayer travel moves are only necessary due to parallel slicing and the current slice not having knowledge of the end XY position of the prior slice. Serial slicing also has the potential to reduce the visibility of the Z Seam as there is no start/stop between layers if the next layer's XY start position flows directly from the previous layer's XY end position.

### Describe the solution you'd like

We believe the OPTION to implement Serial Slicing could solve this problem with the additional option to set the XY start position of the next layer being the same as the XY position of the previous layer.

We envision a typical workflow of using the parallel slicing for various preview trials as we optimize slicer settings due to it's speedy performance. But then (optionally) do a final slice in serial mode. There are certain prints where we would sacrifice slicer time for reduced print time and improved print quality.

### Describe alternatives you've considered

Combing and Z Seam Settings can be used to influence how interlayer travel moves negatively impact print quality, but do not always behave as expected.

Neither approach impacts the extra print time that comes from all of the extra interlayer travel moves.

### Affected users and/or printers

Business like us need to ensure print quality the first time. Having to reprint costs labor time. For most prints, the labor savings of parallel slicing is worth it. But for long or critical prints, an (optional) increase of slicer time upfront is worth the reduction in print time and (more importantly) improved print surface quality and reduction of any z seam.

fvrmr, 2021-12-13:
Hi @mattwn thank you for your feature request. 
Could you explain a bit more why you think this would help with getting less z seam. What is exactly parallel slicing?
Maybe you could visualise your idea.

mattwn, 2021-12-15:
Regarding the question "what is parallel slicing": It is my understanding that the Cura slicer is multithreaded and can slice multiple layers in parallel depending on the number of logical CPU cores.

Since layer N and layer N+1 are sliced simultaneously, layer N+1 has no knowledge of layer N's end position. For this reason, the XY start position of layer N+1 can not be the same as the XY end position of layer N.

Regarding "getting less z seam": if the move from layer N's end position to layer N+1's start position was constrained to the Z axis only (because the XY coordinates of both positions matched), the extrusion could flow continuously (similar to the continuous extrusion "layer change" of spiralize/vase mode), resulting in less dots, stringing, retraction artifacts, etc. 

This can only happen if layer N is sliced BEFORE layer N+1. Which would require the option to set the slicer to "serial" (single threaded). This would have a slicing time cost, but that optional tradeoff is discussed in the original post.

Please let me know if I can provide any more info. Thanks.

Ghostkeeper, 2021-12-23:
For some background, in Cura 2.5 we implemented multi-threaded slicing. This makes slicing big models ~20 times as fast on modern computers (more like ~4x on computers of 2016). But it had the disadvantage that Cura can no longer optimise which part to start printing with on a layer to minimise travel moves from the last part of the previous layer. This is a feature request to reinstate the old behaviour as an option, at the cost of greater slicing time.

Note that this does not influence the seam. The seam is optimised to be closest to a certain position (user-specified) or to prefer corners of different angles (sharpest corner and shortest) or to be spread out (random). Multi-threading is not an issue there because it can be optimised individually for each layer. This only affects which part to start printing with. When printing multiple parts, it has the potential to turn a long travel move into a shorter one.

But honestly, I would rather have a real solution that doesn't require a trade-off. For instance, we could calculate the optimal order of printing parts beforehand sequentially (after the slicing stage and before the planning stage) and then do the rest of the slicing process concurrently.

I think though that we have to defer this issue. The advantage of shortening this travel move is nice, but we have bigger issues to tackle. This is an issue, but we're not giving it any priority unless things change.

mattwn, 2024-02-26:
In case anyone else is interested in this, @smartavionics incorporated this feature into his Cura fork. It was implemented several releases ago, 4.20.14. It works great and cuts down on travels and strings at layer change. 

This in addition to the improved combing in the smartavionics fork has completely eliminated stringing in our workflow. I can't praise his fork enough.

https://github.com/smartavionics/Cura