Ultimaker/Cura #11050: Build Plate Adhesion

Pedrx, 2021-12-10:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Ender 5 +

### Reproduction steps

Attempted to add a Brim

### Actual results

[image]

### Expected results

Not the above.

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

None

GregValiant, 2021-12-10:
Check your Brim Line Count.  At 2000 and a .4 line width that's 800mm wide all around the model.  Your build plate would need to be 1700 x 1700 to fit that.
If you click on the icon next to the number it will re-set back to "Brim Width / Line Width" which (dependent on line width) should be around 20.

Pedrx, 2021-12-10:
Thank you for prompt reply but no, I didn't insert a brim line count of 2000! 
It appears of its own volition and refuses to go away.
Not with the CE3 though. In fact I'm getting a successful print, so far, by using Cura to prepare the file for the CE3 then shoving the SD card in the CE5.

GregValiant, 2021-12-10:
"support_brim_line_count" has a maximum of 50 before the warning shows up and the default appears to be 20.  Those should have loaded when you installed the Ender 5 in Cura.
The Cura folks are going to need a project file (File | Save Project) and probably a log file (Help | Show Configuration Folder and "Cura.log") so they can figure out where the 2000 came from.  Zip them up and post them here.
Did you try switching to "Skirt" or "Raft"?

closed 2022-01-06 by no-response[bot]