Ultimaker/Cura #11053: Controls to Set the Animation Playback Speed

Garryb123, 2021-12-10:
### Is your feature request related to a problem?

I would like to be able to slow down or speed up the playback extrusion animation depending on whether I desire a close inspection of the extrusion path or would like to rush through a section.

Also, some shapes play back at different speeds and the default speed can seem too fast or too slow so being able to alter the speed would be nice.

### Describe the solution you'd like

I'd request the addition of playback speed controls (e.g.,  0.25x   0.5x  0.75x   Normal   1.25x   1.5x   1.75 x  2.0x   4.0x) just above the playback control.

Ideally the playback speed would change immediately in real-time upon clicking the desired speed control so the user could switch between speeds depending on which layer in a model he wants to inspect (or rush through).

[image]

### Describe alternatives you've considered

There don't seem to be alternatives.

### Affected users and/or printers

All users and printers would be affected.

fvrmr, 2021-12-13:
Hi @Garryb123 thank you for another feature request. 
You can use your arrow keys when using simulation view. Let me know if that could help you.

Garryb123, 2021-12-13:
> Hi @Garryb123 thank you for another feature request. You can use your arrow keys when using simulation view. Let me know if that could help you.

Thank you for the suggestion. The arrow keys seem to affect rotating the model or possibly speeding up the animation playback just a little. I'd like more speed control as in the graphic I made above.

It seems others have requested a similar feature:
https://community.ultimaker.com/topic/23453-animation-speed/

fvrmr, 2021-12-17:
Thank you for your reply, I have discussed it with the team and we decided to defer this request. Since it would only have value if the animation was the real print speed and that is not the case. I hope you understand!

Garryb123, 2021-12-17:
> Thank you for your reply, I have discussed it with the team and we decided to defer this request. Since it would only have value if the animation was the real print speed and that is not the case. I hope you understand!

Thank you for considering my request.

Maybe I didn't explain well but my idea was simply to make controls to either speed up, or slow down, the default animation speed when the default animation speed wasn't fast or slow enough to the users liking; there need be no relation to the actual print speed.

But I understand if you wish to defer this. Thanks again and best regards.

Ghostkeeper, 2021-12-23:
An unobtrusive and logical way to implement this would be to speed it up by 5x while holding shift. When holding shift, the arrow keys progress by 5 lines instead of 1, so this is a logical extension that wouldn't need any interface element.

aconz2, 2022-05-08:
Landed here from searching for a want of the same thing. Then I discovered that the keyboard shortcut mentioned above does exist and mostly suited my needs and is mostly documented [here](https://support.ultimaker.com/hc/en-us/articles/360011733980-The-Ultimaker-Cura-workflow-explained) as

> Path slider. This indicates the current position of the simulation view. This handle can be dragged or moved with the ‘arrow keys’ or ‘shift + arrow keys’

And from looking at the code, holding shift will advance by 10. This is a pretty big help on its own. One thing that tripped me up was the rotation as mentioned, but you just have to click on the circle on the slider so that it turns blue, then the keyboard will work.

Interestingly, this was part of the initial implementation from 2017 [here](https://github.com/Ultimaker/Cura/commit/98d48978a860d91457af87430941f57df4292066#diff-e68bdfdd124aebcd8f972ed701f0e440dfeea2236e8089bbdbe93e616dc89180R124-R125).

Maybe a few extra words in the docs would help a search engine expose the feature to more people, because it has been very helpful already!

jagodragon, 2022-12-11:
I was really hoping to find this feature.

To me (and i would assume most people) there is little to no value in simulation of a print if it's going to as long as an actual print.

I have to say your dev team is WAY of mark in their assessment in this area. Coming from an automotive engineering back ground, if our simulation took as long as creating an actual part. we would never use simulations.

the hot option implemented in 2017 seams ok. if it didn't make the simulation stop at the end of each layer.

it would be ideal if we could just have a slide to set simulation playback speed.

fieldOfView, 2023-12-12:
Ha, good timing, bot. Work is currently underway to make the preview play speed more in line with actual (stepper motor) speed. Perhaps this feature could be rolled into that work?

https://github.com/Ultimaker/Cura/pull/17250