Ultimaker/Cura #11054: Material Flow crash

danielbreul, 2021-12-11:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Creality CR 10S Pro

### Reproduction steps

Load the project "Fußhalter.3mf" and Slice

### Actual results

The result looks normal in cura. But when printing, from layer 19 onwards, the first part of a line has too much flow, so that the print then looks like it does in the picture "cura 4.12.1.jpg".

### Expected results

I then installed Cura 4.7.1 again. The problem does not occur with the same settings. See "cura 4.7.1.jpg"

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Fußhalter.zip](https://github.com/Ultimaker/Cura/files/7696083/Fusshalter.zip)
[image]
[image]

GregValiant, 2021-12-11:
Wow.  those are some really aggressive settings.  What layer cooling system do you use?

The 10mm retraction distance may be coming into play as the 10mm prime at the start of a layer happens before the nozzle starts moving.  I would suggest disabling both "Retract at Layer Change" and "Retract Before Outer Wall".

The only change I made before attempting to print was to change the retraction distance to 6.5 (that's the max I'll go with Ender 3 Pro and Micro Swiss hot end).  The flow rate of 10.8mm³/sec is right at the upper end of what my hot end is capable of.  I have a single 5015 blower system and I don't think it was enough at your settings.
At any rate, I would call the dimensional accuracy and the finish on both prints using your settings as not acceptable.  The third print was with my normal print settings at .2 layer, .4 width, 75mm/sec.
Using your settings - 
- The 4.7 print below the floor is 40.44mm and above the floor is 40.75.
- The 4.12.1 print is 40.5 below the floor and 41.00 above.
With my settings:
- 41.03 below the floor and 41.14 above.
In both cases the light hitting the step along with being printed with silky PLA is making it look worse than it is.  It might be similar to the "Benchy Deck" syndrome.

Within your settings, the speeds vary, the "Retract before layer change" and "Retract before outer wall" are having an effect as is the 10mm prime distance and .8 line width (from a .4 nozzle).
Whether this is a bug or not isn't my call.  I do think your settings are having an effect.

The Z seam locations are circled.  The inner Z seam on the 4.7 print doesn't look too good either.  My print is on top, your 4.12.1 print is in the middle and your 4.7 print is on the bottom.
Even on the corners away from the Z seam, with your settings it looks like the pressure in the nozzle is having trouble building up at the start of an extrusion.
[image]
[image]

Ghostkeeper, 2021-12-23:
This looks like a print quality problem. We can't reproduce the print quality since we don't have your printer available. To fix this, we'd need to know what we need to change in Cura's output.

Is there anything you can indicate in Cura's g-code that is going wrong?

We are not here to optimise your printing settings. This issue list is for bugs and feature requests.

closed 2021-12-27 by no-response[bot]