Ultimaker/Cura #11055: Décalage entre la pièce et le Raft

sebr6, 2021-12-11:
### Application Version

4.12.1

### Platform

windows 10

### Printer

Snapmaker A350

### Reproduction steps

Lors du tranchage avec un raft il y a un décalage entre la pièce et le raft . 
[image]

### Actual results

Décalage

### Expected results

Enlever le décalage , imprimer sur le raft

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Uploading S2A_Groot_hole_fixed.zip…]()
[Uploading cura.log…]()

sebr6, 2021-12-11:
[cura.log](https://github.com/Ultimaker/Cura/files/7696924/cura.log)
[Uploading S2A_Groot_hole_fixed.zip…]()

fvrmr, 2021-12-13:
Hi @sebr6 thank you for your report.
I can't open your project file, since it was not done with uploading. But you could check the `Raft Air Gap` setting. 
Also for next time please fill in the form in English.

sebr6, 2021-12-14:
> Hi @sebr6 thank you for your report. I can't open your project file, since it was not done with uploading. But you could check the `Raft Air Gap` setting. Also for next time please fill in the form in English.

After adding and modifying the raft air gap it's ok.
thank you very much 👍

Ghostkeeper, 2021-12-23:
This is the intended behaviour of the Raft Air Gap setting.

closed 2021-12-23 by Ghostkeeper