Ultimaker/Cura #11056: Under Extrusion on 0.2mm Nozzles

Vecotr, 2021-12-11:
### Application Version

4.12.1

### Platform

MacOSX

### Printer

Ender 5

### Reproduction steps

1. Slicing a model (chess figure) with standard 0.2mm nozzle profile (ultra quality, 0.08mm)
2. Export to SD-Card
3. Print and wait

### Actual results

Strong under extrusion when printing smaller diameter circles/cylinders (roughly 5.55mm and smaller).
Other shapes, like cubes (10mmx10mmx10mm) or larger diameter cylinders work just fine and print without any problems.
Tried changing layer-height, temperature and print speed but nothing helped/changed the end result)
Changing the flow parameters wasn't an option since other shapes printed perfectly with the standard settings.
For the same reason, I can also rule out that the extruder motor is not strong enough.
Photo shows the printed part of the end result up to the point where the printer basically printed thin air.

[image]
.
This issue only occurs on 0.2mm nozzle diameter. 0.4mm or even larger one work perfectly.

### Expected results

Normal Print, without under extrusion
[image]

I can't say for sure if it's a bug or a problem with my machine but I checked all possible causes of error and couldn't find one. I also tried multiple different models and found the same problem on every single one.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I hope, that this is helpful.
If it's not and it's just a problem with my machine (or I am an complete idiot) do not hesitate to tell me :)!
Thank you.

fvrmr, 2021-12-13:
Hi @Vecotr thank you for your report. 
If it is happening on multiple models with the same nozzle, it looks like the nozzle is causing this. 
What all possible causes did you check? Did you check if your nozzle is clogged?

Vecotr, 2021-12-14:
Hi Fenne!

Thank you for your quick response.
The nozzle was the first thing I checked.
 I‘ve never printed with a nozzle that small before, that‘s why it‘s a brand new one.
The error occurred right at the very first print I did with it.
I checked for any clogs but when I stopped the print and fed some filament manually (without any cleaning after stopping the print), everything worked fine (both when feeding some filament by hand and by manual control of the extruder-motor).

After that, I did some further investigations. I varied some parameters such as print temperature, layer height and print speed and found out, that none of them changed the situation.

Furthermore I found out, that this problem only occurs when printing circles as small as 5.5 mm in diameter and smaller. 
Imagine you print a cone with a base diameter of let‘s say 15mm (decreasing in diameter the higher up you go). The print would work flawlessly until the diameter of the cone falls below 5.5mm. 
Then the printer starts to lay down far to little material and prints some spongy structure (the extruder motor is calibrated properly).

I‘m not an expert at 3d printing technology but after all the only error I can think of is the way, how the extruder-motor handles the filament flow in this specific case. And that‘s determined by the gcode. 
This is why I reached out to you.

I hope these informations are useful and I‘m very grateful that you are trying to help me out with this. :)

Ghostkeeper, 2021-12-23:
This is a printing quality issue. It's generally something we can't really help you with. We don't have your printer, so we can't reproduce the problem. To fix this, we need to know what Cura is doing wrong. If you don't know what Cura is doing wrong, we also can't help you.

We could determine if there is something wrong with the g-code if you provide a project file (File -> Save Project). What I would look at is if the adjacent lines are indeed adjacent without gaps between them, and if the line widths are correct. You can also see that in layer view, but if you really want to be sure you can calculate the extruded volume for a g-code line, and divide that by the length and the layer height to get the line width.

We cannot help you tune your printing profiles. I fear you might just have to look at [how to fix underextrusion](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/underextrusion.md) online, unless you can indicate that Cura is doing something wrong.

So what should we change in Cura's output to fix this issue? What is Cura doing wrong here?

closed 2022-01-06 by no-response[bot]