Ultimaker/Cura #11057: Preview shows extra layers/extrusion paths not present in generated G-code when slicing multiple objects with Print Sequence set to "One at a time"

trevordyck, 2021-12-12:
### Application Version

4.12.1

### Platform

Windows 10 21H1 (OS Build 19043.1348)

### Printer

Creality Ender 3 Pro

### Reproduction steps

Add two or more _different_ objects to the project and set "Print Sequence" to "One at a time". Slice and look through all layers in Preview.

### Actual results

Look through all layers in Preview for any of the following:

- One or more layers which incorrectly shows all layers being printed, with extra travels to/from 0,0,0. Prior and subsequent layers will appear correct in Preview. These travels are not reflected in the gcode generated by Cura, when analyzed with https://ncviewer.com/
- Extra layers at the end with no travels or extrusions. These travels also not reflected in the gcode generate by Cura.
- Does not appear to affect multiples of just one same object. Sometimes does not affect just 2 or 3 different objects. Does not appear to be linked to a specific STL file that I've tested, or specific combination of STL files.

[image]

[image]

[image]

[print-one-at-a-time_preview-bug.zip](https://github.com/Ultimaker/Cura/files/7697922/print-one-at-a-time_preview-bug.zip)

### Expected results

Preview seems to work fine when using Print in order on multiples of the same STL file, or a single file. Gcode seems to be generated correctly, it's just the preview that seems to be wrong.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Here are a handful of STL's I've been troubleshooting this issue with. The STLs seem to be fine on their own, or in groups of just one STL file, so I don't think it's the STL files that are the issue.

[E3P Accessories and Parts.zip](https://github.com/Ultimaker/Cura/files/7697925/E3P.Accessories.and.Parts.zip)

fvrmr, 2021-12-13:
Hi @trevordyck thank you for your report.
I can reproduce your issue and I will bring it up with the team. Keep you posted!

fvrmr, 2021-12-14:
I have discussed it with the team and I created a ticket on our backlog for this.
Devs see CURA-8762