Ultimaker/Cura #11058: screen Freezes

RMichiel, 2021-12-12:
### Application Version

4.12.1

### Platform

Mac Monterey Version 12.0.1 Graphics Radeon Pro 580 8 GB

### Printer

Creality Ender 3 Pro

### Reproduction steps

I have post processing turned on so that it tells me remaining Tim to completion. Has worked fine on previous versions. This version does not give me the time remains but worse. THE CONTROLS FREEZE. I CAN NOT PUSH TO PAUSE OR CHANGE TEMP. LOCKED IN HOME SCREEN. PERCENTAGES DO NOT RESPOND ON THE SCREEN EITHER. The prints complete ok and then things go back to normal.

### Actual results

If I load the same file with the same specs on the previous version it works fine.

### Expected results

I'm sure this is a bug that hopefully gets corrected. I print almost every day and want to use the new features, but not sacrifice having control of the printer to override or make subtle changes.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No screen shots necessary. I believe I explained the issue.

fvrmr, 2021-12-13:
Hi @RMichiel thank you for your report.
So your Cura screen freezes when using USB and using the monitor view? Or does your printer screen locks up?

RMichiel, 2021-12-13:
I’m using the SD card and it’s the printers screen that locks up

Rich Michiel

Ghostkeeper, 2021-12-23:
The Display Progress On LCD post-processing script puts either `M117` or `M73` commands in the g-code. You could try switching the Time Reporting Method setting to see if your printer handles the other one better.

Neither M117 nor M73 specifies that the screen of the printer should freeze:
* https://reprap.org/wiki/G-code#M117:_Display_Message
* https://reprap.org/wiki/G-code#M73:_Set.2FGet_build_percentage

So if the printer's screen freezes when it encounters this g-code, I would say that is a bug in the printer's firmware, not a bug in Cura. I think you'd have to report this with Creality, not with us.

closed 2021-12-23 by Ghostkeeper