Ultimaker/Cura #11059: Bridging settings have wall expansion that damages print quality

Taudris, 2021-12-12:
### Application Version

4.12.1

### Platform

Windows 11

### Printer

Custom FFF

### Reproduction steps

1. Load a model that has bridges with a width that is not an even multiple of the line width.
2. Enable experimental bridging settings.
3. Configure a lower flow rate and print speed for bridges.

### Actual results

Slice the attached project and go to layer 11.

In "no bridging", everything looks fine.

In "bridging w/ no speed change", you can see the bridge skin overlapping with the walls. This effectively creates one or two bridge lines that have more plastic than the other bridge lines, which can affect the surface quality of the bridge.

In "bridging w/ wall & skin 50% speed" (50% in this case being 60mm/sec), you can see the outer and inner walls grow significantly at one or both ends of the line segment. The growth is proportional to the speed change to some degree, though it appears to have a maximum amount of growth (eg you can't inflate it to insane widths by setting a speed like 20mm/sec). According to @Ghostkeeper, [this is intentional](https://github.com/Ultimaker/Cura/discussions/10931#discussioncomment-1763924). However, on my printer, it results in visible defects.

[image]

### Expected results

* No overlap between non-wall lines and the innermost wall lines.
* No wall growth.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I'm using Klipper with pressure advance enabled. This allows my printer to more accurately reproduce corners and commanded line widths. It therefore causes this aspect of the bridging settings to have a negative impact on print quality, as the wider lines are faithfully reproduced in the final print.

[4010 duct - bridge setting defects.zip](https://github.com/Ultimaker/Cura/files/7610422/4010.duct.-.bridge.setting.defects.zip)

smartavionics, 2021-12-12:
Hi @Taudris , my cura releases that you can find at https://github.com/smartavionics/Cura/releases feature an extra setting that lets you configure the amount of boost applied at the end of the bridge wall lines so you can set that to 0 disable the feature. Unfortunately, that (and many other tweaks/fixes) never made it into UM cura.

Ghostkeeper, 2021-12-23:
These are two issues, really. I'll focus on the skin lines overlapping with the walls. If we want to track multiple issues separately, they should really be separate issues. I'll handle the first one.

This is a more general issue, not particular to bridging. Here is a simple project that reproduces the issue (see the first layer):
[skin_lines_overlap.zip](https://github.com/Ultimaker/Cura/files/7770049/skin_lines_overlap.zip)
I'll see what the rest of the team thinks about it.

As for the second one, that is indeed intentional behaviour. It's just not configurable at the moment.

idmadj, 2023-08-03:
@Ghostkeeper 
"As for the second one, that is indeed intentional behaviour. It's just not configurable at the moment."

Would love to see a setting for this, similar to "Bridge Wall Coasting". Is there any preliminary work on this, otherwise I'd be happy to submit a PR.