Ultimaker/Cura #11060: Weird Layer infill

mydavydiv, 2021-12-12:
### Application Version

4.12.1

### Platform

windows 10

### Printer

ender 3 pro

### Reproduction steps

tried to print original cura file "retraction tower" but i realized that it does the same problem with personal files as well 
[image]

### Actual results

[image]

### Expected results

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

youtube video link 

https://youtu.be/v1VLsamQt5g

fvrmr, 2021-12-13:
Hi @mydavydiv thank you for your report. 
I am not able to reproduce your issue. Could you share a project file (file - save project)?

Ghostkeeper, 2021-12-23:
This looks like a duplicate of https://github.com/Ultimaker/Cura/issues/4576. A project file would clarify that, because the Top Surface Skin Layers setting can't be seen in the video (though other Top Surface Skin settings are visible, so it is probably >0).

closed 2021-12-27 by no-response[bot]