Ultimaker/Cura #11061: Make Overhang Printable - Inside Only

atomicalsoftwares, 2021-12-13:
### Is your feature request related to a problem?

Infill is the most time consuming part in most prints. On seriously large projects infill even with "lightning infill" takes a lot of time!

### Describe the solution you'd like

Printable Overhangs INSIDE ONLY.
Add an option that can select if the printable overhangs are outside only, inside only or both!

### Describe alternatives you've considered

Using lightning infill

### Affected users and/or printers

All

atomicalsoftwares, 2021-12-13:
Just to ensure I've expressed myself correctly:

Make over hang printable adds material below an overhang at a certain degree:
```

--------|   <---Regular overhang

-----|   <- Printable overhang
     /
   /
/
```

It would be great that this option can be selected for the *inside* of the object as well, not just the outside!

fvrmr, 2021-12-13:
Hi @atomicalsoftwares thank you for your request.
We have a setting called `skin edge support thickness` this setting adds an extra line through the infill to support the edge of the skin. Read here more about this setting: https://github.com/Ultimaker/Cura/issues/11048
Let me know if that could also work for you.

atomicalsoftwares, 2021-12-13:
Hello @fvrmr !

Thanks for the quick reply.

I didn't knew about that feature and it'll come in handy but it isn't exactly what I'm hoping for.

Given a large enough model to be printed, it's wall are usually strong enough to support the whole print, therefor, infill is not required.

Take this model for example:
[image]

As it prints, it doesn't need supports or infill.

[image]

Once you start reaching the top, you can add "lightning infill" to support the overhang on the inside.

[image]

What I'm trying to get accomplished is, instead of using lightning infill, that we're able to make the overhang printable by adding material.

For example:

[image]

Here the area in red will be at a 90 degree angle, not easy to print!

But if you add the option "make overhangs printable" this is the result:

[image]

I'd like the software to be able to accomplish this on the outside, inside or both.

This means that enormous prints that can support their structure purely on the walls (because they're strong enough) don't need support / infill on the inside.

Hopefully that'll be clear and if not, just let me know and hopefully I can continue providing feedback to this excellent software.

fvrmr, 2021-12-17:
Hi @atomicalsoftwares thank you for your explanation with pictures! 
I have discussed it with the team and we decided to defer this request. We don't think this would value since `Lightning` is actually building up from the walls. I hope you understand!

atomicalsoftwares, 2021-12-17:
> Hi @atomicalsoftwares thank you for your explanation with pictures! I have discussed it with the team and we decided to defer this request. We don't think this would value since `Lightning` is actually building up from the walls. I hope you understand!

I'm glad it was considered atheist!

One last Keypoint before closing...

Lightning infill is not part of the structure, unlike printable overhangs which modifies the structure itself. Lightning infill is prone to print to a point where it could fail. Printable overhangs do not, because they're part of the structure providing rigidity and support. I would find it interesting to see the time required to make an internal overhang printable vs Lightning infill at diverse percentages. 

Still tho, I will follow an independent solution and maybe in the future it could be considered!

Good luck to you and the team! 

Happy new year and hope 2022 bring us more exiting features!

closed 2021-12-17 by atomicalsoftwares