Ultimaker/Cura #11064: "Brim only on Outside" does not work

normkawakami, 2021-12-13:
### Application Version

4.12.1

### Platform

Windows10

### Printer

Ender3 V2

### Reproduction steps

1. Sliced with CURA 4.12.1 with "Brim only on outside" checked, but inner brim were also made. 
[image]

2. With other model,  "Brim only on outside" was un-checked, but only outside brim was made. 
[image]

### Actual results

1. Inner brims were made.
2. Only outer brim was made.

### Expected results

1. Only outside brim should be made.
[CE3_eagle-Wing5_LW-left-profile3.zip](https://github.com/Ultimaker/Cura/files/7702876/CE3_eagle-Wing5_LW-left-profile3.zip)

2. Inner brims should be made.
[CE3_eagle-Wing6_LW-left-profile3.zip](https://github.com/Ultimaker/Cura/files/7702877/CE3_eagle-Wing6_LW-left-profile3.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3_eagle-Wing5_LW-left-profile3.zip](https://github.com/Ultimaker/Cura/files/7702926/CE3_eagle-Wing5_LW-left-profile3.zip)
[CE3_eagle-Wing6_LW-left-profile3.zip](https://github.com/Ultimaker/Cura/files/7702927/CE3_eagle-Wing6_LW-left-profile3.zip)

fvrmr, 2021-12-14:
Hi @normkawakami thank you for your report. 
Both your files show model errors and probably this is related to the brim only on outside not showing up correctly.
[image]

Because if I slice a model with your settings it is working correctly.
Brim only on outside disabled: 
[image]

Brim only on outside enabled:
[image]

So I think you need to check your models and make sure that there are no model errors.

normkawakami, 2021-12-14:
Hi Fenne,
Thank you for your investigation.
I have understood that the reason is model error. 
I will ask this issue to the model maker.

Norm Kawakami
Japan  

Windows の メール から送信

差出人: Fenne
送信日時: 2021年12月14日 18:09
宛先: Ultimaker/Cura
CC: normkawakami; Mention
件名: Re: [Ultimaker/Cura] "Brim only on Outside" does not work (Issue #11064)

Hi @normkawakami thank you for your report.
Both your files show model errors and probably this is related to the brim only on outside not showing up correctly.

Because if I slice a model with your settings it is working correctly.
Brim only on outside disabled:

Brim only on outside enabled:

So I think you need to check your models and make sure that there are no model errors.

Ghostkeeper, 2021-12-24:
I think the first one is a case of https://github.com/Ultimaker/Cura/issues/4222.

The second one is indeed possibly a model error. It should produce a brim on the inside, but if the model is not manifold, Cura may think that the hollow inside gets filled with infill/skin.

closed 2022-01-07 by no-response[bot]