Ultimaker/Cura #11066: IMPORTANT PROBLEM

mionmichele98, 2021-12-13:
### Application Version

4.12.1

### Platform

win11

### Printer

alfawise u30 pro

### Reproduction steps

I import only the file

### Actual results

I don't know

### Expected results

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2021-12-14:
Hi @mionmichele98 thank you for your report.
Like @fieldOfView mentioned we need more information to help you. Can you share a project file (file - save project) with us?

Asterchades, 2021-12-15:
Honestly just looks like a badly made model being used, which is resulting in a badly formed slice - garbage in, garbage out. Either source a corrected model from the original designer, experiment with the provided mesh fixes in Cura, or correct the model yourself (manually via a mesh editor like Blender or SketchUp, or automatically such as with 3DBuilder).

Ghostkeeper, 2021-12-24:
Or the wall visualisation was ticked off in the layer view legend?

closed 2021-12-28 by no-response[bot]