Ultimaker/Cura #11067: can not multiply object

DanielJutz, 2021-12-13:
### Application Version

4.11.0

### Platform

arch linux

### Printer

jg aurora a5

### Reproduction steps

fresh install on arch linux with default custom FFF printer
- load any stl
- right click on model
- click "multiply selected"
- hit enter

### Actual results

Box "Placing objects" appears, but progress bar never starts and object doesn't get multiplied.

### Expected results

Progress bar completes and duplicate object appears.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[cura.log](https://github.com/Ultimaker/Cura/files/7705725/cura.log)

fvrmr, 2021-12-14:
Hi @DanielJutz thank you for your report. 
Are you using the official AppImage? And does this also happen in 4.12.1?

DanielJutz, 2021-12-14:
I got it from the arch package manager. When using the official AppImage for 4.11.0 or 4.12.1 the issue does not occur.

Ghostkeeper, 2021-12-24:
The log file shows this relevant error:

```
2021-12-13 18:28:18,240 - ERROR - [JobQueueWorker [2]] UM.JobQueue.run [135]: Exception: Job <cura.MultiplyObjectsJob.MultiplyObjectsJob object at 0x7f7d5849ca00> caused an exception on worker JobQueueWorker [2]
2021-12-13 18:28:18,242 - ERROR - [JobQueueWorker [2]] UM.JobQueue.run [135]: Traceback (most recent call last):
2021-12-13 18:28:18,243 - ERROR - [JobQueueWorker [2]] UM.JobQueue.run [135]:   File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 133, in run
2021-12-13 18:28:18,244 - ERROR - [JobQueueWorker [2]] UM.JobQueue.run [135]:     job.run()
2021-12-13 18:28:18,245 - ERROR - [JobQueueWorker [2]] UM.JobQueue.run [135]:   File "/usr/lib/python3.10/site-packages/cura/MultiplyObjectsJob.py", line 70, in run
2021-12-13 18:28:18,246 - ERROR - [JobQueueWorker [2]] UM.JobQueue.run [135]:     found_solution_for_all = arrange(nodes, Application.getInstance().getBuildVolume(), fixed_nodes,
2021-12-13 18:28:18,247 - ERROR - [JobQueueWorker [2]] UM.JobQueue.run [135]:   File "/usr/lib/python3.10/site-packages/cura/Arranging/Nest2DArrange.py", line 126, in arrange
2021-12-13 18:28:18,248 - ERROR - [JobQueueWorker [2]] UM.JobQueue.run [135]:     found_solution_for_all, node_items = findNodePlacement(nodes_to_arrange, build_volume, fixed_nodes, factor)
2021-12-13 18:28:18,248 - ERROR - [JobQueueWorker [2]] UM.JobQueue.run [135]:   File "/usr/lib/python3.10/site-packages/cura/Arranging/Nest2DArrange.py", line 94, in findNodePlacement
2021-12-13 18:28:18,249 - ERROR - [JobQueueWorker [2]] UM.JobQueue.run [135]:     converted_points.append(Point(point[0] * factor, point[1] * factor))
2021-12-13 18:28:18,250 - ERROR - [JobQueueWorker [2]] UM.JobQueue.run [135]: TypeError: Point(): arguments did not match any overloaded call:
2021-12-13 18:28:18,251 - ERROR - [JobQueueWorker [2]] UM.JobQueue.run [135]:   overload 1: argument 1 has unexpected type 'numpy.float64'
2021-12-13 18:28:18,252 - ERROR - [JobQueueWorker [2]] UM.JobQueue.run [135]:   overload 2: argument 1 has unexpected type 'numpy.float64'
```

I think this comes down to bad compilation of pynest2d, perhaps with the wrong Sip version. The `Point` constructor indeed requires `long` arguments, but it should convert automatically, at least with the Sip version we're using at the moment.

I consider this an issue with the Arch package maintainer, not with us. Although we may be able to make it easy on them and maintain other Sip versions (e.g. by adding explicit casts), we currently only support one Sip version. We don't have the manpower to maintain more. Sorry.

closed 2021-12-24 by Ghostkeeper

C0rn3j, 2022-01-03:
I managed to make it here with another/similar Arch issue that is not present in the AppImage, so am just linking the relevant Arch bug report - https://bugs.archlinux.org/task/73160

And Cura issue related to it https://github.com/Ultimaker/Cura/issues/10785

For those who manage to stumble upon this the same way.

Ghostkeeper, 2022-01-13:
As an aside, we intend to update Python to 3.10 and Sip to .25 in the next release. This is currently still on the `CURA-8592_python_3.10` branches. The issue should be fixed then, and probably also fixes it for Arch's distribution.