Ultimaker/Cura #11068: Error en Visualizacion de capaz

Mecadiarg, 2021-12-13:
### Application Version

12.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

A la hora de segmentar la pieza la visualización de las capas  se ve de forma "lineal", intente reinstalar el programa, probe con versiones anteriores y me sucede los mismo.... este problema comnezo cuando actualice Cura a su ultima version, previo a eso no eh tenido nunca problemas.
probe ehjecutarlo como administrador y resolver los problemas de compatibilidad, probe varias opciones y nada funciono

### Actual results

Error en la visualizacion

### Expected results

Arreglaerlo

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2021-12-14:
Hi @Mecadiarg thank you for your report. 
From your screenshot you are in compatibility mode and this is probably because your driver or GPU doesn't support OpenGL 4.1.
You could check in the preferences if the `Force layer view compatibility mode` is enabled, which you can disable. But if it is disabled and you still have compatibility mode and then this is related to your GPU or driver not supporting OpenGL 4.1

I will close this issue since this is something we won't fix.

fvrmr, 2021-12-14:
Also for next time when you report a bug, please fill in the form in English.

AUDISIND, 2024-09-09:
Hola, tengo el mismo problema, habrá actualmente alguna solución para esto