Ultimaker/Cura #11070: Changing infill density in "Modify Settings For Overlap", no longer working in Cura 4.12.1

Dgilla, 2021-12-13:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

S5

### Reproduction steps

1. support blocker
2. Scale and a surround an area on my object
3. Per Model Settings/ Modify settings for Overlap
4. Change infill density to 80%
5. Change Infill pattern to grid
6. Slice
7. Warning pops up, unable to slice
8. Have done this many times before, now not working in the new Cura

### Actual results

Wont slice

### Expected results

Will slice

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[S5-CUIRA-No-Slice.zip](https://github.com/Ultimaker/Cura/files/7707335/S5-CUIRA-No-Slice.zip)

fvrmr, 2021-12-14:
Hi @Dgilla thank you for your report. 
Your model is set to don't support overlaps in the per model settings, so there is nothing to slice. 
Change your model to per model settings - normal model. And then you can slice your model.
[image]

GregValiant, 2021-12-15:
[image]

Dgilla, 2021-12-15:
Great thanks for your help,
It was just that I selected the model and blocks, instead of just the blocks when changing the infill density

closed 2021-12-24 by Ghostkeeper