Ultimaker/Cura #11072: Usable print area and adhesion brims

hamsterdave, 2021-12-14:
### Is your feature request related to a problem?

When a part is close to, but still within the boundaries of the printable area, having an adhesion brim or a skirt enabled that would extend beyond the printable area causes the software to consider the part too big, but for no obvious reason. Because the brim isn't visible prior to slicing, this can lead to confusion as to why the part cannot be printed, and the error message does not specifically call out this possibility. It is a mistake you only make once, but it is a maddening problem until you realize what's going on.

### Describe the solution you'd like

An error message pops up when the Slice button is clicked in this scenario. It currently says:

Please review settings and check if your models:
-Fit within the build volume
-Are assigned to an enabled extruder
-Are not all set as modifier meshes

Adding "No adhesion brim or skirt extends beyond the printable area" to this message would hopefully remind the user of that possibility. It didn't become apparent to me what was happening until I scaled the model down, sliced it, previewed it, and realized the brim was nearly off the build area.

### Describe alternatives you've considered

A more robust solution would be to generate a red approximation of the skirt/brim that appears when the model has the stripes indicating that some part is outside the build area. This would immediately call attention to the specific problem, however it sounds like a lot more work to implement.

### Affected users and/or printers

All users and all printers with a finite bed size, but especially *new* users who haven't made this mistake yet.

Asterchades, 2021-12-15:
It does already have an indicator for the bed adhesion size. If you look at the outer edge of the print bed, even in Prepare mode, the surface is a darker grey - adjusting the size of the bed adhesion (skirt distance, brim width, or raft extra margin) will adjust how far in that darker grey extends.

While there's a small amount of parallax (it's not actually tinting the bed, rather drawing a new layer just above it) you can still use it fairly reliably to see if part of your starting layer extends beyond the remaining usable area.

Ghostkeeper, 2021-12-24:
The expected workflow is that you see your model intersecting with the grey area around the edge. I agree though that it's not well explained why there is a grey area there, and so the link to fix the issue is missing for novice users.

@lorenzoromagnoli Do you have an idea on how we could make this more clear, without adding swaths of text to read?

lorenzoromagnoli, 2021-12-27:
Indeed I also believe that the grey areas in Cura are often interpreted as weird model shadows and that can lead to some confusion. 
Adding a more explicit representation of the adhesion layer,  for instance just showing it in the same colour as the model, could probably improve the user's understanding but it might lead to some inconsistencies.... e.g. While it might be easy to show the brim or raft as part of the model I guess it would be much harder to show any support structure in the same fashion. 

Ideally, if there was a fast way to generate a rough mesh of all things that are going to be printed without actually slicing... I'd love to see those parts in the preparation stage as well. (maybe they could be rendered with a very low opacity something like ghosts).

That would definitely help users understand where, and how to tweak supports, adhesion, etc... 

In general, even if we won't be able to add the visualization for the support structures, and we might create some inconsistencies in the overall behaviour I'm quite keen to work on this proposal and solve an issue that was reported multiple times. Let's discuss this with the all team @Ghostkeeper

Ghostkeeper, 2022-01-03:
We can show the brim as part of the shadow of the model (and we do in one-at-a-time mode) but then you (think you) can't place models such that their brims intersect, which is possible and even advisable in reality. This is why it's around the edge, not around models, unless you are printing in one-at-a-time mode. In one-at-a-time mode the collision of the print head still applies to the brim so that makes sense then.

The shadow can change its colour if we want. Of course we can't change the shadow to be the same colour as the model, because the model can be multiple colours. But it can be the same colour as the adhesion extruder, for instance.

Visualising the support structures before slicing is indeed not very feasible. It would require knowing where the inside of the model is, so at the very least it'd require stitching which requires slicing, or we'd need to move that to 3D operations which is going to be quite complex.
I had an implementation of getting the first layer, just a single slice, in the front-end at some point, using Trimesh which has a function to get cross-sections. This worked nicely, except when loading in big models it would take a few minutes extra, which is not nice. And that's just 1 layer.

fvrmr, 2022-01-11:
We have created a ticket for this on our backlog.
Devs see CURA-8879