Ultimaker/Cura #11073: Slicing failed with unexpected error

dmitrygribenchuk, 2021-12-15:
### Application Version

4.13~202112141335 PPA

### Platform

Linux Mint

### Printer

Creality Ender 3 v2

### Reproduction steps

1) Open STL model
2) Slice it

It's working fine at the previous version. Now it's broken on every model.

### Actual results

Get error: Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker.

### Expected results

Slicing completed

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log.gz](https://github.com/Ultimaker/Cura/files/7718845/cura.log.gz)
[image]

dmitrygribenchuk, 2021-12-15:
Also reproduced on 4.13~202112151335

fvrmr, 2021-12-16:
Hi @dmitrygribenchuk thank you for your report.
Could you also share your project file?

kiss81, 2021-12-17:
I can confirm this issue on the latest ppa builds. It's also complaining that it can't load the following plugins:
Sentrylogger, AMFReader, TrimeshReader

[concolelogcura.txt](https://github.com/Ultimaker/Cura/files/7733503/concolelogcura.txt)
[CCR10_3DBenchy.3mf.txt](https://github.com/Ultimaker/Cura/files/7733506/CCR10_3DBenchy.3mf.txt)

kiss81, 2021-12-21:
I can confirm the slicing issue is solved. I still get the missing plugin errors on
Sentrylogger, AMFReader, TrimeshReader

Ghostkeeper, 2021-12-27:
So this probably was a temporary problem in the version they built nightly. The only change we made to CuraEngine between 2021-12-14 and 2021-12-18 was a fix for tree support which is disabled in Kiss' project file, so I don't know what made it tick there. Could be that those builds are a bit behind and it was https://github.com/Ultimaker/CuraEngine/commit/75d826437aaa18a27c65a18f5620ae960e6511ba.

dmitrygribenchuk, 2022-02-14:
Hello. have the same problem with 4.13~202202021404
[cura.log.zip](https://github.com/Ultimaker/Cura/files/8060445/cura.log.zip)

fvrmr, 2022-02-14:
@dmitrygribenchuk could you share a project file? (file - save project)
Your logs don't show what is causing this.

dmitrygribenchuk, 2022-02-14:
Sure
[CE3_GoPro_Monitor_Mount_Laptop.3mf.zip](https://github.com/Ultimaker/Cura/files/8061830/CE3_GoPro_Monitor_Mount_Laptop.3mf.zip)

fvrmr, 2022-02-22:
Hi @dmitrygribenchuk, thank you it is working for me on master.
Please keep in mind that you are using a nightly we don't maintain. So they are maybe a bit behind with our master branch.

dmitrygribenchuk, 2022-02-28:
Fixed on 4.13~202202271238

closed 2022-03-14 by Ghostkeeper