Ultimaker/Cura #11074: USB Printing

masterchief4115, 2021-12-15:
### Is your feature request related to a problem?

No settings available to user to choose COM ports, Baud Rates or guides on how to do this.
When adding printers only option is network or non-networked, no USB available.
Once added, Cura doesn't detect the printer, even when connected and other software packages can detect and connect.

### Describe the solution you'd like

All the above, but working

### Describe alternatives you've considered

I've considered using Repetier, and do because it works, even though it's a much less welcoming and well-designed software

### Affected users and/or printers

Anet A8

fvrmr, 2021-12-16:
Hi @masterchief4115 thank you for your request. 
We don't maintain USB printing anymore since our printers doesn't work via USB and we don't test 3rd-party printer. 
So that's why I will close this issue since we are not gonna implement this in Cura. 

If you have problems with printer over USB you can check if the USB printing plugin is disabled. Go to marketplace - installed and look for USB printing. Please note that this plugin may interact with other microcontrollers: https://support.ultimaker.com/hc/en-us/articles/4407742254610-Ultimaker-Cura-interacts-with-microcontrollers

closed 2021-12-16 by fvrmr

Ghostkeeper, 2021-12-27:
See also https://github.com/Ultimaker/Cura/issues/5207.