Ultimaker/Cura #11077: Preview errors after Slicing in Arachne Beta 2

charredchar, 2021-12-16:
### Application Version

Arachne Engine Beta 2

### Platform

Windows 10

### Printer

Creality CR-10

### Reproduction steps

Load a model, slice, check preview.

### Actual results

Either the screen gets covered in white or extra moves between layers and 0,0

### Expected results

Preview as normal

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I was running 4.12.0 before Arachne Beta 2, it appeared to have successfully brought over the settings on its own and shows no other issues. Part prints perfectly normal despite the preview errors though it is hard to use the software with this issue.

[image]

[image]

fvrmr, 2021-12-16:
Hi @charredchar thank you for your bug report. 
We are aware of this bug, see https://github.com/Ultimaker/Cura/discussions/10931#discussioncomment-1748963. We have a ticket on our backlog for this.
So I will close this issue and mark it as a duplicate

closed 2021-12-16 by fvrmr

charredchar, 2021-12-19:
> Hi @charredchar thank you for your bug report. We are aware of this bug, see [#10931 (comment)](https://github.com/Ultimaker/Cura/discussions/10931#discussioncomment-1748963). We have a ticket on our backlog for this. So I will close this issue and mark it as a duplicate

I appreciate the link to the discussion but I created an issue as I did not see one previously. The comments mention it is an issue resolved in 4.13 but no mention of Arachne engine. Is there some way to keep track of if/when this gets fixed for Arachne? As of right now, Arachne is useless with this bug.

fieldOfView, 2021-12-19:
It will be fixed in the next Arachne-based release. Whether that is another Arachne beta, or Cura 5.0 beta is as yet unknown. What is known is that the Arachne changed have now been merged into master, so the version of Cura after Cura 4.13 will have the Arachne engine integrated.