Ultimaker/Cura #11078: Certain geometry on first layer is not printed

neildmd, 2021-12-16:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Mingda D2

### Reproduction steps

Opened STL and sliced it with default settings.

### Actual results

Certain geometry on the first layer is not printed.

### Expected results

All geometry should be printed unless otherwise specified (modifiers, etc.).

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

This file slices properly in all previous versions of Cura. I've included a 4.11.0 project using the same settings for comparison.
[4.12.1 first layer glitch.zip](https://github.com/Ultimaker/Cura/files/7724686/4.12.1.first.layer.glitch.zip)

GregValiant, 2021-12-16:
The stringer running from platform to platform is .25mm tall and is 0.10mm below the platforms and it runs from the edge of the right platform, across the middle platform, and to the midpoint of the left platform.  The image below is the part from your 4.12.1 project and shows the feature on the build plate crossing the middle platform.  The part from the 4.11 file looks identical.
[image]

So the three square platforms are not on the build plate in either the 4.12.1 or 4.11 versions.  Changing the initial layer height in 4.11 to 0.1 changes the 4.11 preview to match the 4.12.1 preview.  In 4.12.1 changing the Initial Layer Height to .21 results in a preview like 4.11.
As to why the 4.12.1 version is picking up a 0.1 feature with an initial layer height of 0.2 would seem to be the question..  The 4.12.1 preview looks (to me) to be a more accurate representation of the model.

neildmd, 2021-12-16:
That makes perfect sense now. As you say though, strange that previous versions didn't pick up that feature and 4.12.1 did. So it's parsing the model in finer detail. That's actually a good thing since accurate recreation of the model is what we want to see. Closing thread since it's not a bug at all. Thanks for clearing that up Greg

closed 2021-12-16 by neildmd