Ultimaker/Cura #11080: Optional time or filament usage estimation as suffix/prefix to job filename

tekyinblack, 2021-12-16:
### Is your feature request related to a problem?

No

### Describe the solution you'd like

When needing to schedule multiple jobs, its useful to know the time estimation for each one to say, get the short ones out the way before starting a long one, or just avoiding a long print if a spool is looking a bit low. Having that information on the filename means that its immediately apparent and saves time and mistakes. The format I was thinking of was perhaps
myjobname_HHH-MM.gcode for elapsed time or
myjobname_50g.gcode for filament weight or
myjobname_6m for filament length.

I can add these manually of course but having a default file saving option similar to the machine prefix would be useful and I'm sure be a benefit to many users

### Describe alternatives you've considered

Entering these manually each time

### Affected users and/or printers

All users and printers

Asterchades, 2021-12-17:
All three are already possible via the GCode Filename Format (Plus) plugin(s). Placeholder values exist in this for print time in various format, material length, and material weight, among others, which can be incorporated into a default filename.

Ghostkeeper, 2021-12-29:
Let's merge this with the other feature request that wants to add options to the print job name: https://github.com/Ultimaker/Cura/issues/5368

closed 2021-12-29 by Ghostkeeper