Ultimaker/Cura #11082: Abnormal slice

whoseyoung, 2021-12-17:
### Application Version

4.12.1

### Platform

windows10

### Printer

Geeetech A10

### Reproduction steps

Use the attached slice parameter to preview the slice, the printing effect is very poor

### Actual results

Severe striations appear

### Expected results

Solve the layer pattern problem

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Abnormal slice.zip](https://github.com/Ultimaker/Cura/files/7731571/Abnormal.slice.zip)

GregValiant, 2021-12-17:
I get a "compressed file is invalid" message when I try to extract from that zip file.

fvrmr, 2021-12-22:
Hi @whoseyoung thank you for your report.
Can you share a project file (file - save project) instead of your profile. Also can you show a screenshot what is happening.

whoseyoung, 2021-12-23:
Hi all，
Thank you for your reply. Maybe I made a mistake, and I have solved it just now

fvrmr, 2022-01-11:
Good to know it has been solved. I will close this issue. Feel free to reach out if you have more questions!

closed 2022-01-11 by fvrmr