Ultimaker/Cura #11088: Persistent Axis Indicator Icon

Garryb123, 2021-12-17:
### Is your feature request related to a problem?

I we zoom in on a model and then rotate it we can easily lose track of its orientation as we set slicing parameters, and end up slicing it in an orientation we don't want.

### Describe the solution you'd like

I suggest a permanent axis indicator something like in this image which would always stay visible on the screen no matter the zoom factor. With this indicator we would always know how the part (parts) was (were) oriented at a glance.

[image]

### Describe alternatives you've considered

Zoom all the way out with the orientation buttons but then need to zoom back in on the part to view the nozzle path.

### Affected users and/or printers

All users would benefit from this; none would be hurt.

fvrmr, 2021-12-21:
Hi @Garryb123 thank you again for your feature request.
I have discussed it with our UX designer and he really liked your proposal. So I will bring it up with the team. 
Keep you posted!

Garryb123, 2021-12-22:
> Hi @Garryb123 thank you again for your feature request. I have discussed it with our UX designer and he really liked your proposal. So I will bring it up with the team. Keep you posted!

Thank your for entertaining my request; I'm so happy your UX designer likes this idea! All the best.

Ghostkeeper, 2021-12-29:
I don't think this is really compatible with the way that we also use that indicator to indicate where the origin is. If you have a printer with the origin in the centre of the build plate, the indicator moves to the centre as well.

Garryb123, 2021-12-29:
@Ghostkeeper Where the origin is shouldn't matter. All the icon does is display the scene orientation. The icon always stays at one place on the screen. Here's an example from IronCAD.

https://user-images.githubusercontent.com/5060580/147694310-113564e9-cb8f-4c0d-9cc3-17935bf0de65.mp4

Ghostkeeper, 2021-12-29:
I understand the request, yes. But to some users it does matter where the origin is, I reckon. This is why we have an option to move the origin, after all.

Garryb123, 2021-12-29:
> I understand the request, yes. But to some users it does matter where the origin is, I reckon. This is why we have an option to move the origin, after all.

Thank you for taking time to comment. I still don't understand why there could be an incompatibility because the orientation indicator I suggest has nothing to do with the origin location - they are different things - but will leave it. All the best. :-)

Ghostkeeper, 2021-12-30:
If they are different things, we'd need to implement a new origin indicator. Currently we use the orientation indicator to also indicate the origin, at least.

fvrmr, 2022-01-11:
I totally forgot to get back with the conclusion of this request. 
But we have a ticket on our backlog for this. 
Devs see CURA-8850

Garryb123, 2022-01-11:
> But we have a ticket on our backlog for this.

Thank you!