Ultimaker/Cura #11090: Unable to slice a thin shell correctly

ilko-k, 2021-12-18:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Load the provided .stl file and try to get a decent slice.
[Caliset-Flow_1_perimeter.zip](https://github.com/Ultimaker/Cura/files/7740020/Caliset-Flow_1_perimeter.zip)

### Actual results

[image]

### Expected results

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Even in vase mode, the behavior is bad and the nozzle does a lot of unecessary moves and retractions, leaving gaps and holes. The shell of this object is of 0.4 mm width. I also tried to reduce the line width parameter in Cura with no luck. It seems to occur only when there is a thin wall that is smaller than 0.8 mm.

GregValiant, 2021-12-19:
Line Width .399
Enable Compensate Outer Wall Overlaps.
Vase mode can only be used effectively with models that are solid throughout so that model doesn't qualify.

ilko-k, 2021-12-19:
@GregValiant thanks for the reply but I already tried your suggests and every possible things but no, that is not good at all. SuperSlicer is doing well where Cura fails at this point. Maybe it's not purely a bug but there is definitely something wrong in the algorithm

GregValiant, 2021-12-19:
This is what I get with 4.12.1 with my Ender active.  The .399 line width trick is not something mast new users would think to do.
[image]

This is what I get from Arachne at .4 line width.
[image]

In neither case do they look very much like that first slice you showed.  (This old laptop doesn't support OpenGL 4.1 and so my display is in Compatibility Mode).
If you could do a "File | Save Project" and zip and post the 3mf file it would provide more info.

ilko-k, 2021-12-19:
Just so you know, I'm printing / modeling since years so I'm a little bit experienced. I'm maybe missing something. This is the best I can get but still not good.
[image]

I attached the project file. Thanks in advance for the help.
[Caliset-Flow_1_perimeter.zip](https://github.com/Ultimaker/Cura/files/7740765/Caliset-Flow_1_perimeter.zip)
.

GregValiant, 2021-12-19:
Try changing your Outer Wall Inset to 0 and re-slice.
The model on the left has Outer Wall Inset at 0 and the one on the right is set to .005 in the per model settings.
[image]

ilko-k, 2021-12-19:
oh well, I had never took attention at this particular setting. Now this is what I get :
[image]

It seems fine. Thank you very much for your time.

closed 2021-12-19 by ilko-k

GregValiant, 2021-12-19:
I'm not sure but it seems that like "Remove All Holes" and "Make Overhangs Printable" the Outer Wall Inset appears to be for very specific cases and combinations of settings.  This is the first time I've noticed what it actually does.

ilko-k, 2021-12-19:
hmm interesting, I've never played with these settings before since I did not need any of them. Combining "Remove holes" with regular 0.4 mm line width also gives a good slice result. It adds a few extra internal lines on the uppermost layer of the ground though.
[image]

Ghostkeeper, 2021-12-29:
The printing of small walls was not ideal, but I consider this fixed in Cura 5.0.