Ultimaker/Cura #11093: Flow rate exceeds theoretical maximum about 2 times

akshimassar, 2021-12-19:
### Application Version

4.11.0

### Platform

Arch Linux

### Printer

ZAV printer

### Reproduction steps

1. Open model
2. Set layer height, line width, speed limits, flow percentage
3. Check sliced model flow diagram
4. Calculate theoretical max flow: 0.3*0.6*80 = 14.4 in my case
5. Look out for red zones in the image, layer 9

### Actual results

Flow range varies from zero to 31.87 mm cubic per second.

### Expected results

I would expect max flow rate to be about 14.4

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Most of the model is printed at much lower rate. That makes flow color scheme mostly unusable.
[image]
[project.zip](https://github.com/Ultimaker/Cura/files/7740693/project.zip)

fvrmr, 2021-12-22:
Hi @akshimassar thank you for your report. 
I checked your model in our Arachne build and there the flow rate is as expected.
[image]

I also discussed it with the team and since it is correct in Arachne we don't fix this for the 4.x versions, because Arachne will be our main engine in 5.0.
For now you can use our beta of Arachne: https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta_2

Ghostkeeper, 2021-12-30:
Fixed for Cura 5.0 in Ultimaker/CuraEngine#1210.

closed 2021-12-30 by Ghostkeeper