Ultimaker/Cura #11094: No horizontal slide bar

pm77g, 2021-12-19:
### Application Version

4.12.1

### Platform

Apple OS10.14.16

### Printer

Anycubic

### Reproduction steps

Since I updated to the latest version

### Actual results

I used to have the horizontal slide bar, I don't have it anymore. The vertical slide bar is working fine.

### Expected results

I expect having both horizontal and vertical slide bars

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Here is how I see Cura now, without the bottom slide bar

[https://ibb.co/GR97zt3](url)

fieldOfView, 2021-12-20:
Your Cura is running in "compatibility mode", either because macOS or your GPU driver refuses to give Cura an OpenGl 4.1 context to render to, or because you have selected the "Force layer view compatibility mode (restart required)" in the general preferences of Cura (eg to fix other issues). "Compatibility mode" has limited functionality, and one of the functions that is missing is the horizontal slider.

fvrmr, 2021-12-22:
Hi @pm77g thank you for your report. 
Closing this issue since @fieldOfView is correct and we will not fix this.

closed 2021-12-22 by fvrmr