Ultimaker/Cura #11097: Cannot position objects matching exactly the build plate size

readnotify, 2021-12-20:
### Application Version

4.11

### Platform

Mac

### Printer

Chiron

### Reproduction steps

1. Creat something the same X and Y size of your build plate
2. import it

### Actual results

puts it in a weird spot, and give an error trying to position it
[image]

### Expected results

Should place it in the exact (and only) place it fits.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

cannot upload STL here

GregValiant, 2021-12-20:
Load the model, set Cura up to slice, then use the "File | Save Project" command to create a 3mf file.  Zip the file and post it here.

Asterchades, 2021-12-21:
If you're using a bed adhesion method which extends beyond the model it won't actually fit. You'll either need to use a raft with 0mm expansion or turn off the bed adhesion altogether.

Also, you posted too soon after trying to upload the file. It didn't finish so there's a dead link instead of a picture.

closed 2022-01-03 by no-response[bot]