Ultimaker/Cura #11100: Better Lighting options

imLokiGod, 2021-12-21:
### Is your feature request related to a problem?

I am shocked this has never been addressed, and for my years of using Cura, it has always rubbed me the wrong way. When you are looking at whatever model (in in this example I will use a basic temp tower.), the way the lighting is done on the models, make it near impossible to read lettering or numbers, especially is viewed straight on. It would be nice to be able to control where the light is originating from, or at least the intensity. Having to pan and tilt around to find a proper angle to make any writing or marks on the model visible is frankly a pain. If I load a temp tower, I should be able to read clearly all of the numbers and lettering from any angle. Honestly have no idea why this has never been addressed 

### Describe the solution you'd like

Lighting options. Being able to adjust where the light is originating from, and intensity. To allow full and uniform lighting of the model and having details be clear with any view or angle. 

### Describe alternatives you've considered

As far as I see, there are no alternatives for this problem currently 

### Affected users and/or printers

This would benefit anyone using Cura

fvrmr, 2021-12-23:
Hi @imLokiGod thank you for your request. 
I will bring it up for discussion with the team. Keep you posted!

Ghostkeeper, 2021-12-30:
Need to put yourself in the shoes of a less technical user here. Not a lot of users would even realise that there is a light or even that it would be necessary to have light to view a 3D model. I don't think having options to adjust the light position or intensity would solve the problem completely, because to someone who hasn't worked with 3D rendering it's not something they would think about. They wouldn't know it's a solution.

And some people inevitably end up reporting bugs about the model being black because the ambient light is toned way down. Or to not see any depth because the ambient light is toned down. Or that a lot of material colours appear lighter or darker than the real spools.

I'd like to pose some alternatives instead:
* Add a second light source at an angle to draw specular reflections for. Maybe a third to get the classic 3 point lighting set-up.
* Implement ambient occlusion.
* Move the current light source to be at a greater angle away from the camera centreline.

Of those, I think ambient occlusion would be the most effective, but also the hardest to implement.

Anyhow, I think this is a duplicate of https://github.com/Ultimaker/Cura/issues/5976.

closed 2021-12-30 by Ghostkeeper