Ultimaker/Cura #11101: Layer height in G-code and Cura profile layer height do not match.

NNGB-kor, 2021-12-21:
### Application Version

4.12.1

### Platform

Windows 11

### Printer

Sapphire Plus

### Reproduction steps

I assemble the 0.3mm nozzle and set the profile layer height of Cura to 0.16.

### Actual results

In the G-code content, the Z-axis is stacked by 0.2 mm.

### Expected results

I thought the Z-axis in G-code would increase in multiples of 0.16.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Is this a bug?
Or is this the intended setting?

NNGB-kor, 2021-12-21:
If the layer height is entered as 0.16mm, but the first layer height is set to 0.2mm, the Z-axis of G-Code continues to increase the Z-axis in multiples of 0.2mm.
I manually corrected the height of the first layer to 0.16mm, and as a result, it was confirmed that the Z-axis of the G-Code was normally stacked in multiples of 0.16mm.

GregValiant, 2021-12-21:
Are you sure you have the order correct?  The first box under Quality is Layer Height and below that is Initial Layer Height.  Initial Layer Height is always the height of LAYER:0 in the gcode.  Moving up in the Z - the rest of the Layers will be "Layer Height" tall.  I think the only thing that affects the Layer Height setting is if you have enabled "Use Adaptive Layers" in the Experimental section.
In your example of "Initial Layer Height = 0.2mm" and the "Layer Height = 0.16mm" then you should see the height of Layer:0 = .20, Layer:1 at .36, Layer:2 at 0.52, etc.  That is what I get now with Cura 4.12.1 and with all previous versions of Cura.

NNGB-kor, 2021-12-21:
> Are you sure you have the order correct? The first box under Quality is Layer Height and below that is Initial Layer Height. Initial Layer Height is always the height of LAYER:0 in the gcode. Moving [...]

I thought it would work that way too. 0.2 and 0.36 0.52.. However, as a result of opening and checking the g code file, the Z value increased by a multiple of 0.2 from beginning to end.

GregValiant, 2021-12-22:
"adaptive_layer_height_enabled":
                {
                    "label": "Use Adaptive Layers",
                    "description": "Adaptive layers computes the layer heights depending on the shape of the model.",
                    "type": "bool",
                    "default_value": false,

So it is supposed to be disabled by default.  What printer definition do you use for the Sapphire?

closed 2022-01-05 by no-response[bot]