Ultimaker/Cura #11102: Add Sovol SV04 profile

aerkiaga, 2021-12-21:
### Is your feature request related to a problem?

Sovol has put a new 3d printer on sale this year, the SV04.

### Describe the solution you'd like

Since the previous models have their respective profiles already part of Cura, it only makes sense that SV04 also be added by default.

### Describe alternatives you've considered

Their webpage also offers a Cura profile for download [here](https://sovol3d.com/pages/download), labeled 'Sovol SV04 Cura Setting( Mac)' (*sic*), which one can use with no updtream modifications to Cura. It's still confusing that it reads 'Mac' in the link, 'Windows& Mac' in the file itself, when it should work on all platforms, or that the download page is titled 'Firmware Download'...

### Affected users and/or printers

It will benefit SV04 users, especially those that are less technical and/or confused by the aforementioned labels. Plus the profile could be improved based on experience with the new machine by members of the Cura community and updated to keep up with new Cura features.

### Additional information & file uploads

[Here](https://drive.google.com/file/d/17RH_DxwnvT5tDZ5cEmAei68C5FvBgkQW/view) is the official profile provided by Sovol. According to user reviews, it works with satisfactory results. I would simply add and push it, unchanged, to the Cura repo, but I'm not sure that's the proper way to add support to a printer, plus there could be intellectual property issues (they don't seem to include a license and I don't know if those rights apply in this case).

fvrmr, 2021-12-22:
Hi @aerkiaga thank you for your request.
We don't add these profiles ourselves. But you can create a pull request with these profiles yourself! 
Read here how: https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura

I will close this issue, but reach out if you have questions!

closed 2021-12-22 by fvrmr

aerkiaga, 2021-12-23:
I have the intention to do so. Maybe I'll experiment a bit with the printer before, I want to make sure the profile works as expected...

Novafpv, 2022-10-28:
> ### Describe alternatives you've considered
> Their webpage also offers a Cura profile for download [here](https://sovol3d.com/pages/download), labeled 'Sovol SV04 Cura Setting( Mac)' (_sic_), which [...]

Actually you need put those files here:
"Contents>MacOS>share>cura>resources>"

NOT initial "Contests>Resources>share>cura>resources>"