Ultimaker/Cura #11103: Setting Panel

Vaibhavkhakkar, 2021-12-21:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Creality Ender 5

### Reproduction steps

Setting box not expanding. 

### Actual results

Setting box not expanding. 

### Expected results

Setting box not expanding. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Setting box not expanding.

fvrmr, 2021-12-22:
Hi @Vaibhavkhakkar thank you for your report.
Could you maybe share a screenshot/video of what is happening? 
And could you also share your log file? You can find your log file here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

Ghostkeeper, 2021-12-30:
Duplicate of https://github.com/Ultimaker/Cura/issues/5527.

closed 2021-12-30 by Ghostkeeper