Ultimaker/Cura #11104: Cura 4.11.0 changing extruder temperature on its own

stevemayathome, 2021-12-21:
### Application Version

4.11.0

### Platform

Windows 10 Pro

### Printer

creality ender 3 pro

### Reproduction steps

after slicing, saving, then printing, pieces were coming out not well printed. Sags, stringing etc. Layers not adhering.
[M104 changing text.txt](https://github.com/Ultimaker/Cura/files/7758955/M104.changing.text.txt)
 

### Actual results

I checked out machine settings, everything checks out ok.
So I edited the sliced file and noticed it was changing the extruder temperature quite wildly.
after changing all m104 settings the same the prints came out ok.

### Expected results

[M104 changing text.txt](https://github.com/Ultimaker/Cura/files/7758936/M104.changing.text.txt)
[After Repair M104 changing text.txt](https://github.com/Ultimaker/Cura/files/7758946/After.Repair.M104.changing.text.txt)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I originally noticed that my printers temperature was changing on the display and had thought my extruder was wonky. Then after diving deeper found this. I do hope this is where I should be reporting this. If it isn't then I apologize.
Thanks for any information you can give me
[cura.log](https://github.com/Ultimaker/Cura/files/7758963/cura.log)
.

GregValiant, 2021-12-22:
It says it was post-processed.  Did you happen to print a temp tower or some other calibration model?
Next to the slice button - is there an icon with a red circle?  That would indicate that a post-processor is active (maybe ChangeAtZ?)

stevemayathome, 2021-12-22:
Hi Greg;

 

Yes there is, I hadn’t noticed it. Now to figure out how to turn it off.

 

Thanks for the speedy and accurate reply.

 

Steve

stevemayathome, 2021-12-22:
Hi Greg;

 

Don’t worry, I turned it off. Wont happen again.

 

Have a Merry Christmas

 

Steve

closed 2021-12-30 by Ghostkeeper