Ultimaker/Cura #11106: preview is only showing a gray ghosted outline of part not the layers

zachts, 2021-12-22:
### Application Version

4.12 and 4.13

### Platform

Windows 10

### Printer

ender 3 pro

### Reproduction steps

4.11 still works however I was working on slicing a file in 4.12 and switched from layer height color coding back to line type and the preview now no longer works in either version. tried uninstalling, rebooting and reinstalling both 4.12 and 4.13 and the problem persists. see attached screen shot.  It's possible this is just a setting I cant seem to find, I was on my laptop and using a mouse with the laptop keyboard and may have made some errant mouse clicks with the touch pad.(I always forget to disable it when using a mouse, it's quite annoying sometimes)  any help in fixing this bug is much appreciated. thanks. attached logs from versions 4.11, 4.12, 4.13 no project file included, doesn't seem to matter, I've tried dozens.
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/7759709/cura.log)
[cura.log](https://github.com/Ultimaker/Cura/files/7759705/cura.log)
[cura.log](https://github.com/Ultimaker/Cura/files/7759706/cura.log)

### Actual results

does this with all files I try to slice now.

### Expected results

should see the layer preview.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7759715/cura.log)
[cura.log](https://github.com/Ultimaker/Cura/files/7759716/cura.log)
[cura.log](https://github.com/Ultimaker/Cura/files/7759718/cura.log)
[image]

zachts, 2021-12-22:
doh, never mind, a friend told me I unchecked the Extruder 1 box in the menu.........dang it, don't know how I didn't see that.  incidentally why would it be useful to turn that off when you only have one extruder to preview?

closed 2021-12-22 by zachts

Ghostkeeper, 2021-12-30:
It wouldn't be. Better not turn it off.

It's useful to be able to turn them off e.g. to hide a model that's obscuring a part printed with a different object, or to hide things that will dissolve away with PVA, etc.