Ultimaker/Cura #11110: Issue with Tree and Normal support

StuartP34, 2021-12-22:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Anycubic Vyper

### Reproduction steps

Trying to build a Covid Virus planter (image attached). Has failed everytime with support.

Print at .1 or .15mm

Tried Tree support using "everywhere" or "buildplate", doesn't matter. 

Tried Normal everywhere and buildplate

### Actual results

When using Tree support the 2nd level of Virus arms (proteins) does not have enough Tree structure underneath to adhere and build on. Multiple types of PLA, always fails at this same place. The extruder starts off building the protien knob on top of the Tree support as a small kidney shaped object. But there is not enough Tree structure to hold it or build on and it falls off.  If the Tree support was sideways or T-shaped and not in-line it would probably work (it's like a seesaw now). 

"Normal" support also fails on the 1st level of print. If selecting "ZigZag" the zigzag structure does not adhere well to the heated 60deg plate, and using glue, temps, etc do not help. At least one of the 1st layer zigzag supports comes up off the buildplate on the 2nd pass of the extruder. You are criss-crossing strings onto the buildplate and this just does not work well. It would be better for the 1st later was solid and then build the zigzag structure on top of this. if there is a setting to do this I don't know how.

It seems Prusa slicer does work with minimal configuration, but I really like the Tree support, it is so easy to remove with minimal clean-up/damage.

I can send the stl file or gcode, but neither file can be uploaded to this message.

### Expected results

it should finish printing

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

fvrmr, 2021-12-23:
Hi @StuartP34 thank you for your report. 
Could you share a project file (file - save project) and share it in a zip here on Github?

Ghostkeeper, 2021-12-31:
I know and understand this issue. We have a case that reproduces it in this PR: https://github.com/Ultimaker/CuraEngine/pull/1205

We've fixed an issue related to this for 4.13 stable (not yet in the beta): https://github.com/Ultimaker/CuraEngine/pull/1544 In this issue, tree support was not properly supporting all places that are overhang. I expect that in 4.13, this will be much improved with tree support. Most likely good enough for this model.

However there is a deeper issue where some models don't really have a lot of overhang, if they have sharp points pointed downwards. A proposed solution was [here](https://github.com/Ultimaker/CuraEngine/pull/1205) but that had other issues. So for now we'll have to leave this open as a known issue, I think. I'll ask my colleagues what they think we can do about it.