Ultimaker/Cura #11111: Marketplace window can not be sized down if it's 'unlocked'

BloodyRain2k, 2021-12-22:
### Application Version

4.12

### Platform

Win 7 x64

### Printer

Not related

### Reproduction steps

I've been changing the QML file for the Marketplace window for a while now for each version, simply to unlock it's fixed size because it's way too small in my opinion.

[image]

Here are the changes I made. I at first tried limiting the width to 2x the original width, but when testing if that worked I noticed that it can ONLY be sized bigger, but not smaller afterwards.

At one point I got it stuck in resizing so it went across both of my monitors and somehow caused it to crash from that.

### Actual results

I've done the same modification (minus the new maximumWidth value) in the past and so far I don't recall this having been an issue.

### Expected results

It would be nice if someone can take a look as to why it can only be sized bigger but not smaller again, and maybe also increase the default size to more than what it's currently, if it has to stay locked because the underlaying problem is not easy to fix.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

fieldOfView, 2021-12-23:
The whole Marketplace is being rewritten for the next major release, so I don't think this issue will get much attention from devs. It is not an issue with Cura, but with your understanding of QML.

The `minimumWidth` is set to be equal to the `width`. In QML this is not a one-time assignment, but it creates a "binding" between the two values. IE: the `minimumWidth` is always the current `width`. In order to not create a binding, you could set the `minimumWidth` to `UM.Theme.getSize("large_popup_dialog").width` instead of `width`.

BloodyRain2k, 2021-12-23:
I see, I'll try out that, thanks.

Ghostkeeper, 2021-12-31:
The new marketplace design can indeed be made bigger. Not smaller than its original size, but we rigged it better this time so that things scale along.