Ultimaker/Cura #11112: Save a Single Setting to the Profile

Garryb123, 2021-12-22:
### Is your feature request related to a problem?

Sometimes when testing settings we may not want to commit all open changes to the profile BUT we might wish to commit SOME to the profile.

### Describe the solution you'd like

I suggest an 'S' (Save) beside each changed setting that, if clicked, would save only that particular setting to the profile without saving other open changed settings. If the changed setting was undone, the 'S' would disappear.

[image]

### Describe alternatives you've considered

The are none.

### Affected users and/or printers

All users and printers will be affected.

fvrmr, 2021-12-23:
Hi @Garryb123 thank you for your request.
It is a bit familiar to your other request about saving profiles: https://github.com/Ultimaker/Cura/issues/11025
Why would this be better or do you want this to be an addition of that request?

fieldOfView, 2021-12-23:
It is quite different. This request is per setting, as opposed to the other request being for all changed settings.

Personally I would prefer this to be implemented in the context menu of the setting instead of adding another button.

Garryb123, 2021-12-24:
> Hi @Garryb123 thank you for your request. It is a bit familiar to your other request about saving profiles: #11025 Why would this be better or do you want this to be an addition of that request?

Sorry if I wasn't clear. This request is separate from my others but I envision them all working together, and, at the same time.

My idea is to be able to save any change made within a setting by itself to the exclusion of others. For example, say I changed three options within infill. Each option would display the 'revert' button AND a new 'save' or 'commit' button. As soon as I decide I am happy with one option I can click 'save' beside it and save that particular change to the profile without saving the other two changes; they remain unsaved.

This would be so much faster than saving the entire profile and has the benefit of letting the user play with more settings (they remain open and visible) before saving them to the profile (at which time they cannot be seen any longer).

Ghostkeeper, 2021-12-31:
I think it's a good idea, though indeed better left to the context menu. That part of the interface is cluttered enough as it is, and this is quite expert-level profile management stuff.

Garryb123, 2022-01-11:
> We have discussed it with the team and with our UX designer and decided to defer this request. Since we believe it wouldn't add much value to our users. I hope you understand.

Thank you for considering; I understand. Best regards.

closed 2023-07-13 by github-actions[bot]