Ultimaker/Cura #11113: Select and move model with mouse

leonealex, 2021-12-22:
### Application Version

4.12.1

### Platform

Mac OS 12.0.1

### Printer

Tevo Tornado

### Reproduction steps

1. Import Model
2. Try to select model to move it
3. impossibile to select with mouse

### Actual results

nothing.
I cannot select and move the part

### Expected results

select the parts and move the parts

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7765377/cura.log)
[TT_Pistachio_Opener_Better.3mf.zip](https://github.com/Ultimaker/Cura/files/7765379/TT_Pistachio_Opener_Better.3mf.zip)

fieldOfView, 2021-12-23:
If you have multiple displays, you could try the following:
Go to the General Preferences and *uncheck* the option to "Restore window position on start". Restart Cura.

alfonder, 2021-12-24:
Thanks, that helps in my case.

Ghostkeeper, 2021-12-31:
Most likely this is a similar issue to https://github.com/Ultimaker/Cura/issues/8745 but for MacOS 12 (Monterey). We currently can't support MacOS12 yet because Cura's biggest dependency, Qt, doesn't support it yet. Qt is responsible for drawing the 3D scene and handling clicks so that's probably why it won't work.

Currently [not even the most recent version of Qt supports Monterey](https://doc.qt.io/qt-6/supported-platforms.html). And we're currently still on 5.15, the latest LTS until last autumn.