Ultimaker/Cura #11114: Closing the program when opening a file

7lad, 2021-12-23:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Good afternoon. When you open any STL file, the program closes

[cura.log](https://github.com/Ultimaker/Cura/files/7768183/cura.log)

### Actual results

.

### Expected results

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://user-images.githubusercontent.com/53573531/147220713-6d3cc365-41ab-49f2-b2ba-18bb6d2f1c25.mp4

fieldOfView, 2021-12-23:
The problem is that your temp folder (which is inside the C:\Users\[your name] folder) contains non-ascii characters. The easiest woraround at the moment is to create a new user in Windows which uses only ascii ("latin") characters. This issues has been reported before, and will be (has been?) fixed by updating a third party module that Cura uses.

fvrmr, 2021-12-23:
Hi @7lad thank you for your report. 
Like @fieldOfView mentioned this is a duplicate of https://github.com/Ultimaker/Cura/issues/9644 an other work around can be found here: https://support.ultimaker.com/hc/en-us/articles/4407746556690-Ultimaker-Cura-crashes-on-Windows-with-username-that-contains-non-Latin-characters
I will close this ticket since it is a duplicate.

closed 2021-12-23 by fvrmr