Ultimaker/Cura #11119: Cura ignores overlap settings

reneroth, 2021-12-23:
### Application Version

4.12.1

### Platform

Mac OS

### Printer

Ender 5 Pro

### Reproduction steps

1. loaded model into Cura
2. loaded second model to act as overlap setting modifier
3. adjusted infill density setting for the overlap

### Actual results

Cura ignores the setting modifier

### Expected results

overlapping parts should have the changed infill density

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This happens with all settings, they all get ignored, no matter if I choose "Infill mesh only" or "Cutting mesh". Tried different models for setting the overlap too, made no difference.

[image]

GregValiant, 2021-12-24:
It works for me with Infill Mesh and two STL's.  Save a project file, Zip it and post it.
[image]

reneroth, 2021-12-25:
Amended the ticket title to add important information I just found out - this only happens if the print sequence is set to "One at a Time", at which point only the blocks created by the support blocker will have any effect.

Thank you for looking into this, and Merry Christmas!

[overlap-bug.3mf.zip](https://github.com/Ultimaker/Cura/files/7775644/overlap-bug.3mf.zip)

fieldOfView, 2021-12-25:
The workaround is to group the model and the modifier.

fieldOfView, 2021-12-25:
Meshes created by the support blocker tool are "parented" to the mesh you add them to. So CuraEngine sees that they belong together.

reneroth, 2021-12-25:
> The workaround is to group the model and the modifier.

Thank you! This helps tremendously - still think it's a bug so keeping this ticket open :) 

Also just using the opportunity to thank you for the wealth of Cura plugins you've provided us with <3

Ghostkeeper, 2021-12-31:
Duplicate of https://github.com/Ultimaker/Cura/issues/9219.

closed 2021-12-31 by Ghostkeeper