Ultimaker/Cura #11120: QOI format for image preview

oliof, 2021-12-23:
### Is your feature request related to a problem?

the standard PNG image format for slice previews requires a large amount of RAM to decompress which limits the ability to implement this on some low RAM touch devices. 

### Describe the solution you'd like

Supporting an option to render previews in [QOI](https://github.com/phoboslab/qoi)  format would allow for rendering in low RAM environments.

### Describe alternatives you've considered

In low RAM environments the alternative is to continue not having slice image previews.

### Affected users and/or printers

Users using the PanelDue display may benefit from this.

### Additional information & file uploads

Ref: https://forum.duet3d.com/topic/26505/does-anyone-here-work-on-superslicer

Ghostkeeper, 2021-12-30:
I think [PNGDec](https://github.com/bitbank2/PNGdec) is a much more friendly option for PanelDue. It requires even less memory and the PNG format actually has more than a month of battle-testing behind it. There's a decent chance that this image format gets forgotten as "that nice one from December 2021" like most new image formats. But of course, it's not up to us to decide for them. We just need to try to integrate with all of these printers and their formats where feasible.

Since this is only going to be relevant to support certain printers (if those printers start needing it) and we can't test with those printers, we're not likely going to be implementing this at Ultimaker. What's more, Ultimaker itself will not allow us time to implement this since they don't benefit from it either. So we'll have to defer this. It's not going to get any priority, though it would be nice to have (as long as it doesn't increase complexity of code maintenance too much).

oliof, 2021-12-30:
The maintainer of the Duet-RRF plugin for Cura has signalled interest to implement this (as part of their plugin), so I will close this request and keep the discussion to the Duet forums. Thanks for the consideration and the detailed answer.

closed 2021-12-30 by oliof

Kriechi, 2021-12-30:
Initial proof of concept using QOI is now available here: https://github.com/Kriechi/Cura-DuetRRFPlugin/tree/next

dc42, 2021-12-30:
@Ghostkewpwe, I'm sorry, your response "It requires even less memory" is very far from the truth. PNG format uses zlib DEFLATE compression, which needs a 32K sliding window buffer to perform the decoding. The PNGDec docs themselves state that PNGDec needs a MCU with at least 48kb RAM. That's why we can't use PNG format with these displays. Also, PNGdec is around 3000 lines of code, compared to around 300 lines to decode QOI format.

Ghostkeeper, 2021-12-31:
Deflate allows sliding window sizes of less than 32K as well.

QOI can be supported via a plug-in. It's a bit difficult to package since you'd need to package a build for Windows, Linux and MacOS, and then select which library to load dynamically from Python. For an example, you can look at my SVGToolpathReader which does something similar with Freetype: https://github.com/Ghostkeeper/SVGToolpathReader
I hope that helps!

Kriechi, 2021-12-31:
@Ghostkeeper FYI: there is a transpiled Python version: https://github.com/pfusik/qoi-ci/blob/master/transpiled/QOI.py
237 LOC of surprisingly readable Python code with 0 other dependencies!

Ghostkeeper, 2022-01-11:
That certainly makes it easier to make a plug-in!