Ultimaker/Cura #11121: Hole horizontal epansion causing artifacts

Estnx, 2021-12-23:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Sliced one of my projects. 
GCODE was not as I expected.
Some artifacts on model surfaces.
Stl exported from freeCad, but .STL is ok

[image]

### Actual results

[image]

### Expected results

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Laana RGB kaelarihm V1.0 Alumine.zip](https://github.com/Ultimaker/Cura/files/7771102/Laana.RGB.kaelarihm.V1.0.Alumine.zip)
[CE3 Laana RGB kaelarihm V1.2 Alumine BL S50 0.2 1h32m PLA 205C 50C.zip](https://github.com/Ultimaker/Cura/files/7771111/CE3.Laana.RGB.kaelarihm.V1.2.Alumine.BL.S50.0.2.1h32m.PLA.205C.50C.zip)

GregValiant, 2021-12-24:
The large rectangular feature in the middle is a hole and HHE applies to it just like it does to the small screw holes.  When the slicer hits one of the horizontal cross holes, then the large feature isn't a hole anymore and so Hole Horizontal Expansion instantly doesn't apply.  As the layers go up then the big center feature turns into a hole again and HHE applies again.  That's why you see those steps.  If you only want HHE to apply to the screw holes then try doing it with a support blocker configured to change the HHE.

Ghostkeeper, 2021-12-31:
Yes, this is expected behaviour. The hollow inside of your tray is considered a hole just the same as any other gap in a layer.

We could consider making this more configurable, e.g. a filter for the size of holes. However we also don't want to go too deep into these settings. They make it very complex for a user to understand. For instance, if we have a "maximum size hole to expand" setting, a user could be wondering why Hole Horizontal Expansion doesn't work on their model (not knowing about that setting). Cura has a lot of difficult caveats like this, and we'd like to reduce them as much as possible.

So I think we'll just have to offload such features to CAD. If you want some holes to be bigger but not others, you'll have to modify the model to make those holes bigger. Cura is no CAD program, so its model editing is limited.

closed 2021-12-31 by Ghostkeeper