Ultimaker/Cura #11122: Add build layer surface to stack objects

ciphersson, 2021-12-23:
### Is your feature request related to a problem?

Not necessarily a problem, but it is a bit frustrating in the sense you have your settings dialed in. Then you need to print a lot of small oddly shaped parts, and you fill the build area up quickly. Then you have to lay out the next set of parts and create a hole new gcode file. What if you could just have one gcode file for it all according to your build dimensions?

### Describe the solution you'd like

Here is the idea that I believe would be a killer feature. 

What if there was a feature where once you filled up the entire surface area with parts, you had a button you could click that says something like "Add build surface layer?" That intern would automagically add what would amount to a raft surface to the highest Z point of the last surface. I would wager that surface would need supports also. 
So then you have an entire new visual surface to lay parts on, adjust their position, orientation, infill type and supports. 

The raft I mentioned doesn't necessarily have to be "real." It could just be a visual indicator of your new surface layer, you are working with.  I could see this being an issue so as not to waste plastic. If an algorithm on the back end decided it was best to just create supports to hold up the next layer. If that wasn't possible, create a raft for every new surface added. This way, people could take full advantage of the entire build volume in... 3D. 

Joule from 3DprintingNerds did a similar thing manually, [making bulk face shields](https://www.youtube.com/watch?v=UeKfZ5r9s9s&t=0s). Awesome what he explained and taught, but IRL a pain to do every time. Could you imagine having an easy button to click that added a layer, and then you filled your entire build volume with multiple layers? That would be a game changer. If you could change infill types per layer, an even bigger game changer. 

An example would be robots. When 3D printing robot parts, they tend to have a lot of parts. Robot arm's for example tend to take two or three separate prints. What if you had one print job with two, three or 5 layers?  

However... I do think that once sliced, one of those warnings that pop up in the middle of Cura would be needed after you exceed 350 meters of filament. Also in the settings, allow and option to keep build plate heated if you have a filament run out sensor.  

### Describe alternatives you've considered

Manually creating layers. Less than ideal. Time-consuming. 

### Affected users and/or printers

I think everyone would benefit from this. 

### Additional information & file uploads

I want to thank all the developers for making Cura so great. There are a lot of options out there for slicing, but we all know Cura is the default go to, It is the standard that all others are measured by.  That is only due to the great dev's. I sincerely appreciate all the work you all do. 

I included a YouTube video of me explaining this idea here: [Add build layer surface option.](https://youtu.be/r3QLdpKEcqE)

Ghostkeeper, 2021-12-31:
Joel's example there adds a layer of air gap though. An air gap works if you have a model that is perfectly stackable, but not for most models. If you'd want a "raft" (1 or multiple layers of material) you'd have to print something to support that raft. So you'd basically fill up your entire build volume with support. And I think if you're doing that, why not just forego the "raft" layer and simply print the rest of the models hanging in the air, suspended on support? You could even use support interface, if you want the nicer surface area of a raft.

The problem remains though (and you'll need that for your solution too) that you'd want the arranger to arrange stuff higher up too, instead of tossing models next to the build plate.

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant