Ultimaker/Cura #11123: Z hop top layer during ironing process to increase quality of ironing.

jxer, 2021-12-23:
### Is your feature request related to a problem?

no problem, just like it as an enhancement

### Describe the solution you'd like

when selecting iron options, allow a zhop check box, where we can add zhop to the top layer and even the second to top layer so, when ironing, there are no drag marks which can affect quality of ironing (shows up a lot when embedding text on the flat layer)

make it selectable if you want 1-2-3 etc layers of zhop

### Describe alternatives you've considered

turning z hop on for the entire print, is a waste, I only need it for some ironing situations.

### Affected users and/or printers

everyone. shows up PETG pretty easily.

Ghostkeeper, 2021-12-31:
The Combing Mode "Not On Outer Surfaces" is pretty much this. It doesn't comb on the highest and lowest layer, which then forces a retraction and (if enabled) a Z-hop. It's a new option in Cura 4.12. Try it out! Let me know if it's not what you had in mind.

closed 2021-12-31 by Ghostkeeper

jxer, 2021-12-31:
> The Combing Mode "Not On Outer Surfaces" is pretty much this. It doesn't comb on the highest and lowest layer, which then forces a retraction and (if enabled) a Z-hop. It's a new option in Cura 4.12. Try it out! Let me know if it's not what you had in mind.

i will look at combing, but if zhop is enabled won't it zhop through the whole thing?

is there also a better document on the different modes of combing? your help article doesn't explain the different modes.

and google is showing the default is combing all, but not according to 4.12.1 which is not in skin.

i guess my point is , I don't want anything to change, my prints are perfect, is changing to not on outer surfaces going to change the entire print? I just want a check box below ironing that says zhop top layer.

Ghostkeeper, 2021-12-31:
There is no way to enable Z hops on certain parts only and retract for other parts, indeed. Retraction and Z hop are linked for the most part.

A more extensive explanation is provided by the Settings Guide plug-in, if you need it: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/travel/retraction_combing.md

If we were to make Z hops configurable to hop for certain parts of the print only, we'd have to consider also how that would work for [all of the other situations](https://github.com/Ultimaker/CuraEngine/blob/master/docs/assets/retraction_combing_hop.svg). This can become a bit complicated. For instance, if combing is enabled and the combed travel move crosses walls, and Z hop is enabled but starts or ends in an ironed area, does it then still use the combed path?

jxer, 2021-12-31:
ok i am learning, reading your combing guide. what should the default be for combing? it appears from upgrading its still "not in skin" but your article speaks as to how it really should be not in outer layers? little clarity?

jxer, 2021-12-31:
So maybe you can explain, and I appreciate the help so far

going from off to -Within Infill, the combing settings get more strict, so Not on Outer Surface, is actually looser than the settings of Not in Skin, I assume Not in Skin also avoids  highest and lowest layer? which appears to be the default

If this is the case, then this was enabled when I did ironing. of a top surface layer with a monotonic order and Iron.

based on this with no Z hop. it was forced to do a retraction to get to the middle. but the print head still marked it.

so zhop is required.

I'm not sure why you considered this a hard logic gate (no disrespect here, i do limited programming)

I looked at your flow chart, but it doesn't speak about the iron routine in it, the zhop function ask should be added in the ironing flow chart as you do have a check box for "iron" only highest layer" right there, add a zhop fuction. maybe I am missing something, but i don't see why this should affect all other situations.

I would tell it to use the combed path, but does it matter? if the nozzle is already away from the bed, then the path really doesn't matter does it?

jxer, 2022-01-02:
@Ghostkeeper  any comments?

Ghostkeeper, 2022-01-11:
> ok i am learning, reading your combing guide. what should the default be for combing? it appears from upgrading its still "not in skin" but your article speaks as to how it really should be not in outer layers? little clarity?

The default can be different per printer, per nozzle, per material and per quality level. We do have a global default though for printers/profiles that didn't bother to optimise it. The global default is currently ["All"](https://github.com/Ultimaker/Cura/blob/5eee9958a026cd1191ef134a305c3aa87b6a3a59/resources/definitions/fdmprinter.def.json#L4009). If it's different for your printer, that means that whoever created profiles for your printer thought that that value is better for this printer.

> going from off to -Within Infill, the combing settings get more strict, so Not on Outer Surface, is actually looser than the settings of Not in Skin, I assume Not in Skin also avoids highest and lowest layer? which appears to be the default

There is not really a clear ordering from strict to loose. But I think you understood it well enough. "Off" means it's not combing anywhere and would always retract instead (if allowed by the maximum retraction count). "All" means that it does comb, as long as a path exists that avoids going through the outer wall. "Not in skin" means that it does comb, as long as a path exists that avoids going through the outer wall and the skin. "Within Infill" means that it does comb, as long as a path exists that avoids going through all of the walls and the skin. "Not on Outer Surface" means that it does comb, as long as a path exists that avoids going through the outer wall and the top/bottom layer of the print. So assuming the top/bottom layers consist only of skin and outer walls, an ordering could be made like this:

    All > Not in Outer Shell > Not in Skin > Within Infill > Off

Where "All" will allow successful combing most of the time (causing a travel move through potentially visible areas), and "Off" will never allow combing and do the safer retraction+zhop instead.

However this is usually not the case, because Not in Skin can travel through inner walls, even on the highest layer. That is a major difference since most skin/infill patterns have endpoints on the inner walls.

> I'm not sure why you considered this a hard logic gate (no disrespect here, i do limited programming)

If you meant the hard logic gate of Z hopping and retracting together: To keep it simple for the user, basically. I understand that some people might see cases where they want to hop but not retract. Or they might see cases in their print where they want to retract/hop, where in a different area of the print they don't want it. But keeping them linked is a way to make the behaviour more predictable for the user. As you can see, the behaviour is complex enough as it is. Adding functionality to filter Z hops or retractions in more cases only makes that problem worse.

> I looked at your flow chart, but it doesn't speak about the iron routine in it

True. It doesn't consider whether or not it's ironing. One might conclude that if the user applies ironing, they want the surface to be smooth. And you might conclude that to make the surface smoother you'd want to Z hop over those surfaces. However Z hop is not just a tool to make the surface pretty. It can also make the surface worse, through oozing and stringing. We've learned to not assume too much about what effects a change to the path can have on aspects of the print quality. Because all printers are different, all materials are different, all nozzles are different and all combinations of those are different still. There's trade-offs to make between configurability and complexity, so where we can make assumptions we make them, but where not we don't. This is one of the hardest parts about developing a slicer I think. We prefer to leave these optimisations up to the user or the profile makers.
Combinations of settings can be made in profiles though. When someone creates a profile for a specific printer set-up, they are testing with that set-up so they can see if the combination works well or if Z hop actually does make it worse for them.

So I'd say that any profile that enables ironing probably wants the combing mode to be set at least to Not on Outer Surface. Maybe they want it to be Within Infill or Not In Skin. But maybe they do have a different printer that doesn't have a fast enough Z stage and Z hops make it worse, so maybe their profile doesn't have the combination. Cura doesn't make that assumption for them because the profiles can test those assumptions and can be shipped with Cura if they are found to hold true.

> based on this with no Z hop. it was forced to do a retraction to get to the middle. but the print head still marked it.
> 
> so zhop is required.
> [...]
> I would tell it to use the combed path, but does it matter? if the nozzle is already away from the bed, then the path really doesn't matter does it?

Z hop is only done when retracting. But it can still retract without Z hops. You're right in that when it Z hops it never follows the combing path. In the flow chart you'll find no outcome that has "Hop" and "Combing" together. If the nozzle is away from the print, it doesn't need to make detours for combing and can just travel directly.

I decided earlier that "Not on Outer Surface" and "Not in Ironing" are sufficiently similar that we don't need to make a distinction between them. "Not in Ironing" would avoid hitting the outer walls and the ironed parts. The ironed parts are all of the top surface. So the difference is then the bottom surface, usually only 1 layer that the user probably wants to be smooth anyway. We could make that distinction for the bottom layer, but in the interest of the aforementioned simplicity, we won't.