Ultimaker/Cura #11125: Printer profiles do not show up if there is another printer with the same manufacturer but different letters are capitalized

MicobyteMichael, 2021-12-23:
### Application Version

4.11

### Platform

Windows 10

### Printer

N/A

### Reproduction steps

When adding a custom machine definition to the `resources/definitions folder`, if there is another machine definition that has the same manufacturer, if the two printers use different capitalizations of the manufacturer's name (i.e. FlashForge vs Flashforge), then the printers that have the manufacturer's name capitalized differently do not show up in the "add non-networked printers" UI.
I was trying to use [Eugr's FlashForge Creator Pro profile](https://github.com/eugr/Flashforge-for-Cura) to set up this printer. I installed it according to the steps on that GitHub page (copy-pasting the files into the Cura configurations folder, installing the plugin, and restarting Cura), but the printer wasn't showing up. After about half an hour of pulling my hair out, I realized that, in the list of printers, there was a Flashforge Dreamer printer. Upon looking more closely, I noticed that it spelled FlashForge with the second F lower-case. On a whim, I figured "why not, I might as well try to see if that would fix it", and lo and behold, I edited the config file to spell FlashForge with the second F lower-case, and *boom*, my Creator Pro suddenly appeared.

### Actual results

If the printer's manufacturer's name is capitalized differently from another printer with the same manufacturer, then it doesn't show up in the "Add non-networked printer" UI. When the capitalization is edited in the definition file to match the other printer, the printer appears.

Original definition file:
[image]

Printer does not show up:
[image]

Made the F lowercase:
[image]

Printer shows up in the list now:
[image]

### Expected results

The printer should have showed up in the "Add non-networked printers" UI regardless of the capitalization differences of the manufacturer's name.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Log file, in case it helps in any way: (I already `grep`'d it for anything that might be relevant, such as "FlashForge", "Creator", "Pro", etc. and found nothing, but just in case it helps somehow.)
[cura.log](https://github.com/Ultimaker/Cura/files/7771780/cura.log)

rburema, 2021-12-30:
I'll put it on the list to discuss with colleagues. We'll keep you informed.

MicobyteMichael, 2021-12-30:
Thank you, please feel free to let me know if there is anything else I can do to help, or any other details I can provide.

commit 5eee9958 (2022-01-11): Case-insensitive visibility checking for printers

In Qt's ListModel, the section headers collect items case-insensitively. However, when checking whether a printer should be visible (when the section header is expanded) was still checked case-sensitively. As a result, it was hiding all printers that did not use the same casing in its manufacturer name, e.g. FlashForge vs. Flashforge.

Fixes #11125.
closed 2022-01-11 by Ghostkeeper