Ultimaker/Cura #11126: The Printer Setting menu window covers the build-plate image

RaminShalom, 2021-12-24:
### Application Version

4.13

### Platform

Windows 11

### Printer

CRX

### Reproduction steps

Updated to Cura 4.13

### Actual results

The Printer Setting menu window covers the build-plate image

### Expected results

The build-plate image should shrink and not be covered by the Printer Setting menu window

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The build-plate image should shrink and not be covered by the Printer Setting menu window

rburema, 2021-12-30:
Please take a look at this plugin: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/SidebarGUIPlugin
You can install that (without loggin in) in the Marketplace.

When it comes to 'vanilla' Cura without any plugins: This is as designed since 4.0, so not a bug. You can collapse the menu to show the whole build-plate, or shrink or move the menu (double click the header to return it to it to its original position). The build-plate itself can also be moved, rotated, etc.

That said, we know it was never a very popular change, and in the future, we might change it to conform to more peoples expectations. Probably not anytime soon though (not for 5.0 as far as I know for example).

closed 2021-12-30 by rburema

RaminShalom, 2021-12-30:
Thank You,

 

For Your Information, this plugin does NOT work with the Cura 4.13 Beta.

 

Any plans to add the Non-planar slicing into Cura?

 

Respectfully,

Ramin Ohebshalom

fieldOfView, 2021-12-30:
> For Your Information, this plugin does NOT work with the Cura 4.13 Beta.

The plugin was updated to wotk with Cura 4.13 yesterday.

rburema, 2021-12-30:
@RaminShalom 

> Any plans to add the Non-planar slicing into Cura?

We're interested, but nothing has been planned yet. It's something we'd like to have eventually as far as I understand, but again, not anytime soon.

@fieldOfView I have said the magic word, and you where summoned it seems.

RaminShalom, 2021-12-30:
Thank You,

 

Healthy Happy 2022

RaminShalom, 2021-12-30:
What is the difference between Cura 4.13 and Cura Master?

fieldOfView, 2021-12-31:
The master branch is where new features are added. When a new beta is launched, a version-branch is split off from the master branch (4.13 being the latest of these). Once that is split off, no new features are added to the version branch, only bug fixes.

Currently there's a huge difference between master and 4.13, because the Arachne engine has been merged into master. So logically the next release after Cura 4.13 will include the Arachne engine. From what we've heard here, that will be Cura 5.0.

In short: Cura master is somewhere between Cura 4.13 and Cura 5.0.

All of this is quite off-topic from your original issue report though. Conversations like this are more appropriate for the Community Forum at https://community.ultimaker.com, and even there it is appreciated to keep a thread somewhat on topic.

RaminShalom, 2021-12-31:
Thank You,Healthy Happy 2022