Ultimaker/Cura #11127: Overlap shell generation duplicating walls

FredericNk, 2021-12-24:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

1. Load two models 
2. Define one model as overlap
3. Change overlap setting to "cutting mesh"
4. Add "wall thickness" to settings
5. Change wall thickness from base setting
5. Slice model

### Actual results

Walls are generated on both sides of the overlap model e.g. base model has 3 wall lines, in overlap 1 wall line is defined --> overlap has 4 wall lines. 

### Expected results

Wall lines should be connected and infill not separated by extra wall lines.
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[UMS5_example.zip](https://github.com/Ultimaker/Cura/files/7774671/UMS5_example.zip)

BloodyRain2k, 2021-12-30:
That's sadly "as intended" as far as I remember, because as far as Cura is concerned are you literally cutting that part out.

The real issue comes from the fact that most settings won't properly, if at all, /without/ being set to cutting mesh : (

rburema, 2021-12-30:
Currently, this is expected behaviour. When you change from 'Infill Only' to 'Cutting Mesh', it will treat the two sub-meshes as if they are separate meshes for all intents and purposes.

With the wall thickness, we calculate the wall-counts that should be used for both shapes. If we where to implement this, then a single shape does no longer have a set wall-count, which we rely on in many places throughout the engine. Even if we decide to fix this, it will be very difficult (in the sense that we'll have to rewrite large parts).

I can see how this is counter-intuitive in this case.

FredericNk, 2021-12-30:
Thank you very much for the reply and clarification ! 

I found a work around for my specific use case: 
1. Reducing the global wall count setting 
2. Using a mesh which includes everything except my region of interest
3. Using the "infill only" mode and increasing the wall count back to what I want for the global setting

fvrmr, 2022-01-04:
Hi @FredericNk we have discussed it with the team and we decided to defer this issue, like @rburema mentioned it would be very difficult to fix this and is not a priority for us right now.
Also nice to hear that you have found a workaround.  
I will reopen this issue but mark it as deferred.