Ultimaker/Cura #11128: Huge gaps in the center of concentric bottom layers

Hobelschmied, 2021-12-24:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Tevo Tornado

### Reproduction steps

1. Insert the .stl model of a bowl into Cura
2. Slice it with my settings for silk PLA with a 1.0mm nozzle
- Concentric
- 1.0 mm Line Width
- 0.5 mm Layer Height
- 100% Initial Line Width
- Spiralize Outer Contour
- Smooth Spiralized Contours
- 6 Wall Line Count
- 12 Bottom Layers
- etc. (I suppose you can see it in the project file)

### Actual results

The center of the sliced model is showing large gaps between layer lines, while the layer lines around it are all nice and close together.

Changing the nozzle diameter, line width and initial layer line width doesn't fix it, there are always gaps between the inner rings.

When printing, the lines are indeed separated, leaving gaps almost as large as the line width between the inner rings of the bottom layers.

### Expected results

The bottom of the model is completely solid, so it should also print solid.
Prints I have done with these settings before and also with a 0.4 mm nozzle and line width did not have these gaps between layer lines.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[TT_Faceted_Bowl_2.zip](https://github.com/Ultimaker/Cura/files/7774930/TT_Faceted_Bowl_2.zip)
[Cura_layer_gaps.zip](https://github.com/Ultimaker/Cura/files/7774932/Cura_layer_gaps.zip)

Here is the model that I'm trying to print:
[https://www.thingiverse.com/thing:2230232](https://www.thingiverse.com/thing:2230232)

rburema, 2021-12-30:
This is fixed in the upcoming 5.0/Arachne (not in 4.13 yet). You can try to slice with the early preview we have out though: https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta_2  (YMMV, bugs may be present. etc.)

Ghostkeeper, 2021-12-31:
Fixed in Ultimaker/CuraEngine#1210.

closed 2021-12-31 by Ghostkeeper